LibreCAD/LibreCAD #1514: Radius=0 is invalid error displayed repeatedly when zooming if division expression used for radius

jbergengh, 2022-01-20:
The error message **radius=0 is invalid** is displayed in the Command line box repeatedly when using the Center + Radius tool, and entering a division expression for the radius

### Expected behavior
Error should not be displayed

### Observed behavior
Error is displayed repeatedly, when placing a circle and when zooming.

### Steps to reproduce or sample file
Start LibreCAD
Select the Circle, Center,Radius tool
For the radius, enter 57/2, or any other mathematical expression with a division expression.  100/3, 22/7, etc
Zoom in our out using mouse wheel, or place multiple circles on the drawing
Observe in Command Line box:  error message radius=0 is invalid is displayed repeatedly

### Operating System and LibreCAD version info
Mint 20.3
LC 2.2

[image]

jbergengh, 2022-01-20:
Update:  Same condition exists when using the **2 Points, Radius** tool

pollywinters, 2022-01-24:
Error is in the librecad/src/ui/forms/qg_circleoptions.cpp file, function QG_CircleOptions::setAction().
Fix is to change line
            action->setRadius(sr.toDouble());
to read
            action->setRadius(RS_Math::eval(sr));
ie. to use the eval method to evaluate arbitrary math expression, instead of only accepting a numeric expression.
This method is already used in the QG_CircleOptions::updateRadius() function later in the same source file.
Still gives the "radius=0 is invalid" message any time that the Radius entry box is cleared to blank, but otherwise is okay.

Note: Total newbie to working on librecad here (though too many years experience working as a software developer on embedded systems lol) so apologies if not following required procedures for posting problem fixes here. Have been looking at librecad source code to gain some experience with QT. Hoping to soon get the hang of submitting via git and pull requests.

pollywinters, 2022-01-24:
Further to this, I think that the maths expression evaluation method is only used within the option value entry boxes, eg. Radius for circles, Length for lines, it is not used when entering position values on the command line. IMHO it would be rather useful if maths expressions could also be processed during command line entry of coord values, allowing entry of positions such as 
@25.7*2,38.1/3
Instead I frequently find myself using either the "cal" command or an external calculator to work out my required X and Y values, then having to cut/paste or retype the numbers into the command line.
- My intention to look at this as the next bit of librecad source code to understand...
(Probably I should have posted this as a new issue or feature request?)

jbergengh, 2022-01-24:
Thanks @pollywinters That fix gets rid of most  of the error messages.  It still throws one while typing in the expression, right after the / but before the denominator.

You can enter maths expressions in the command line. Try Line 2-points, enter 0,0 for the first point and your example of @25.7*2,38.1/3 for the second point in the command line.

pollywinters, 2022-02-03:
Improved the error messages to get rid of message repeat when zooming with scroll wheel. Message now updates only on change of radius value, and informs reason for error (radius negative, radius zero, or invalid formula). 
Similar change for radius entry on "Tangential 2 Circles, Radius" operation. Also added status message output for this operation when there is no valid solution for the selected circles with the given radius value.
Commits are in pollywinters/LibreCAD fork if anyone is interested! Tags 2.2.0-pw2, 2.2.0-pw3

pollywinters, 2022-03-28:
Pull request created
#1531

closed 2024-05-18 by dxli