LibreCAD/LibreCAD #1518: Unable to exit fullscreen mode after closing a file

rshamsnejad, 2022-02-03:
### Initial situation

I opened a file in LibreCAD, then went into fullscreen mode (`Ctrl+Shift+F`) to look at it. I then closed it, and tried to exit fullscreen mode.

### Expected behavior

I should be able to exit fullscreen mode.

### Observed behavior

`Ctrl+Shift+F` does not work, and the clickable option in `View > Fullscreen` is grayed out. I have to create or open a file for either to work.

### Steps to reproduce or sample file

1. Open a file in LibreCAD
2. Go into fullscreen mode
3. Close the file
4. Try to exit fullscreen mode

### Operating System and LibreCAD version info

```
Version: 2.1.3
Compiler: GNU GCC 11.1.0
Compiled on: Jun 10 2021
Qt Version: 5.15.2
Boost Version: 1.76.0
System: Arch Linux
```

```
$ inxi -bz
System:
  Kernel: 5.16.2-arch1-1 x86_64 bits: 64 Desktop: KDE Plasma 5.23.5
    Distro: Arch Linux
Machine:
  Type: Laptop System: LENOVO product: 42982XG v: ThinkPad X220 Tablet
    serial: <superuser required>
  Mobo: LENOVO model: 42982XG serial: <superuser required> UEFI: LENOVO
    v: 8DET76WW (1.46 ) date: 06/21/2018
Battery:
  ID-1: BAT0 charge: 29.6 Wh (65.9%) condition: 44.9/57.7 Wh (77.8%)
CPU:
  Info: dual core Intel Core i5-2520M [MT MCP] speed (MHz): avg: 797
    min/max: 800/3200
Graphics:
  Device-1: Intel 2nd Generation Core Processor Family Integrated Graphics
    driver: i915 v: kernel
  Device-2: Chicony Lenovo Integrated Camera (0.3MP) type: USB driver: N/A
  Display: x11 server: X.org 1.21.1.3 driver: loaded: intel
    unloaded: modesetting resolution: <missing: xdpyinfo>
  Message: Unable to show advanced data. Required tool glxinfo missing.
Network:
  Device-1: Intel 82579LM Gigabit Network driver: e1000e
  Device-2: Realtek RTL8188CE 802.11b/g/n WiFi Adapter driver: rtl8192ce
Drives:
  Local Storage: total: 350.23 GiB used: 68.81 GiB (19.6%)
Info:
  Processes: 212 Uptime: 22m Memory: 3.72 GiB used: 1.61 GiB (43.4%)
  Shell: fish inxi: 3.3.12
```

lordofbikes, 2022-02-03:
You may try the LibreCAD assigned key `F11` instead to toggle full screen mode.

rshamsnejad, 2022-02-03:
@lordofbikes I did try `F11`, but it wasn't working.

I have now tried the same steps on my Windows machine, and the bug still occurs (but with `F11` instead this time indeed).

```
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 
```
```
Edition	Windows 10 Pro
Version	21H1
Installed on	‎26/‎06/‎2020
OS build	19043.1466
Experience	Windows Feature Experience Pack 120.2212.3920.0
```