LibreCAD/LibreCAD #1519: No tooltips shown for widgets on the 'Categories' toolbar

pollywinters, 2022-02-03:
### Expected behavior
Mouse hover over tool widgets shows tooltip text, this works for most of the standard toolbars. But not for the items on the 'Categories' toolbar.

### Observed behavior
Mouse hover over the widget icons for the tool categories - Lines, Circles, Freehand, Ellipses, Polylines, Select, Dimensions, Modify, Measure - does not show a tooltip.

### Steps to reproduce or sample file
Hover mouse over most any widget icon on the LibreCAD window and a tooltip will appear. But not when hovering over the categories icons as listed above.

### Operating System and LibreCAD version info
Linux Mint 20.3, LibreCAD 2.2.0-rc3

### Solution
Tooltips can be added by insertion of "setToolTip()" calls within the librecad/src/ui/lc_widgetfactory.cpp source file, function LC_WidgetFactory::createCategoriesToolbar().
Eg. 
>    tool_button = new QToolButton;
>    tool_button->setPopupMode(QToolButton::InstantPopup);
>    tool_button->setIcon(QIcon(":/icons/line.svg"));
>    tool_button->setToolTip(tr("Lines"));      **// this line inserted**
>    categories_toolbar->addWidget(tool_button);
>    tool_button->addActions(line_actions);

### Source committed on fork
Commit is in pollywinters/LibreCAD fork if anyone is interested! Tag 2.2.0-pw1

fa201, 2022-02-04:
Thank you for the issue and the solution.

ryanthara, 2022-02-15:
> Thank you for the issue and the solution.

And where is the Pull Request for it?

pollywinters, 2022-02-15:
Erm... First time that I've done any change to be considered for inclusion,
and I'm not yet sure how the pull request system works. I'll read up on the
site and try to figure it out.
Best regards, Polly

pollywinters, 2022-03-28:
Pull request created
https://github.com/LibreCAD/LibreCAD/pull/1530

commit f334bdff (2022-06-07): Merge pull request #1530 from pollywinters/issue_1519

no tooltips shown for widgets on the 'Categories' toolbar, fixes #1519
closed 2022-06-07 by lordofbikes