LibreCAD/LibreCAD #1520: Compilation error on dimangular.cpp

araujo88, 2022-02-07:
Steps followed:

 - git clone from souce (success)
 - qmake -r (success)
 - make

Error output:

```
lib/engine/rs_dimangular.cpp: In member function ‘virtual QString RS_DimAngular::getMeasuredLabel()’:
lib/engine/rs_dimangular.cpp:144:42: error: ‘QStringLiteral’ was not declared in this scope
     int dimaunit {getGraphicVariableInt( QStringLiteral( "$DIMAUNIT"), 0)};
                                          ^~~~~~~~~~~~~~
lib/engine/rs_dimangular.cpp:144:42: note: suggested alternative: ‘QStringList’
     int dimaunit {getGraphicVariableInt( QStringLiteral( "$DIMAUNIT"), 0)};
                                          ^~~~~~~~~~~~~~
                                          QStringList
```

g++ version:
`g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0`

OS: 
`Ubuntu 18.04.6 LTS`

fa201, 2022-02-07:
Did you try with: `make -j4` ?

araujo88, 2022-02-07:
> Did you try with: `make -j4` ?

yes

araujo88, 2022-02-07:
Apparently, this was an issue due to Qt4 being installed alongisde Qt5. I completely uninstalled it:

`sudo apt-get purge qt4-designer qt4-dev-tools qt4-doc qt4-linguist-tools qt4-qmake`

Now it works! Thanks

fa201, 2022-02-07:
@lordofbikes Does this mean that we cannot compile if 2 versions of Qt are installed on the system ?

lordofbikes, 2022-02-08:
@fa201 no, it's a matter of correct environment.
Qt Creator handle this with settings.
When using make from terminal/command line, there are batch files on Windows and qtchooser on Linux to set the environment for the needed Qt version