LibreCAD/LibreCAD #1521: Need to release LibreCAD's nightly builds atleast monthly

ajk2196, 2022-02-16:
Please do this as we need to test the newly implemented features and be able to use them if a solution has been found. Guys like me with no experience in coding are just sitting there, watching their problems solved but waiting to be implemented; checking more than once daily; and believe me, everytime I check it, I am always disappointed that the LibreCAD version with new features hasn't been launched yet.

I humbly request for this; I know there are many tasks on the plate, but this is something that will keep things hot. We aren't even having as many issues; the colder our rate of encountering newly raised issues gets, the colder will the users' interests become.

On a very important side note and still having in mind the fact that we have some tasks at hand, let me tell you that I have tried to become an avid user of LibreCAD, and I have raised all the issues that a mediocre-architect-on-the-way-to-become-an-expert will face when using LibreCAD. Of all of them, few have had good responses, while the rest have barely received any attention. Few of them have been solved too and I am much thankful to this community (especially @melwyncarlo) for that. If possible, please give my issues a look and see if the priorities can be changed to have them solved first. Hope @melwyncarlo is in good health.

fa201, 2022-02-16:
I guess you use Windows... Did you check this on sourceforge ?
https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/

jbergengh, 2022-02-16:
@lordofbikes, if there is anything more that I, as an unofficial-beta-tester can do to help get some of these fixes incorporated, please let me know.  I use LibreCAD on a daily basis.  I have cloned and merged in several of @melwyncarlo's fixes and have tested them quite extensively.  My favorites are listed below.  These are the ones I have been using daily for quite a while now, and have found to be stable, if that helps.
Issue 1396 (Task_4)
Issue 1301 (Task_15)
Issue 1351 (Task_31)
Issue 1444 (Task_33)
Issue 1452 (Task_40)
Issue 1458 (Task_43)

ajk2196, 2022-02-17:
@fa201 , yes I have seen the page you have linked to. The last release is of 10th date ofJanuary. There have been many solved issuesand implemented features since then which need to be integrated into a newer release and that release be launched as soon as possible as a nightly build.

ajk2196, 2022-02-17:
Also, please consider implementing these solved issues in the next nightly build, if possible:

1477, 1480, 1482, 1483

gerardbm, 2022-02-17:
I don't understand why there are so many pull requests not merged.
I can understand why programmers stop helping in LibreCAD code.

dxli, 2024-05-18:
We have latest development continuous build for the master branch now.

closed 2024-05-18 by dxli