LibreCAD/LibreCAD #1524: Intersection between 2 circles missed Snap Intersection

lucabalsanelli, 2022-03-01:
I'm drawing two circles intersecting at two points. I want to join the points with a line using the "2 Points" line-tool. "Snap Intersection" places the end points of the line more or less 1cm (over a 1m) off the intersection.

### Expected behavior

Line starting and ending precisely at the intersection between the two circles.

### Observed behavior

Line is starting and ending off the intersection points.

### Steps to reproduce or sample file

1) Draw two intersecting circles
2) Snap to intersection
3) Line tool to connect the two intersection points
4) Zoom in at the intersections

### Operating System and LibreCAD version info

Version: 2.2.0-rc3-23-g3c4163a9
Compiler: GNU GCC 11.2.0
Compiled on: Feb 20 2022
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

lordofbikes, 2022-03-01:
This is just an issue with visual representation of the circles.
Try this:
1. use Circle tool **2 Points, Radius** with radius 500
2. click 0,0 and 500,0 and move mouse up to create the upper circle
3. click 0,0 and 500,0 and move mouse down to create the lower circle
4. the circles now intersect at 0,0 and 500,0 by design
5. switch all snaps off, except _Snap on Intersection_ and create the line
6. zoom on the line ends

You will see, that the line exactly ends on the intersection coordinates.
But the circles don't cross these points they were created on.
I measured the distance from line ends to visual intersection with about 0.1, but this depends on angles and is much less than 1cm.

You can do some math and check circle centers (use _Properties_ tool or _List Entities_ plugin) to see, that the math is correct.

I have some vague ideas and imaginations about the cause, but I can't say where this offset exactly comes from nor what we should do to eliminate it. Important is, that the math is correct.
Finally we're talking about 0.02% (100 / 500 radius * 0.1) tolerance and I see no solution with reasonable expense.

A possible solution would be to write an own, more precise rendering engine. Or maybe find a third party library. But what I expect is, that its performance will be much more worse than currently. So the results won't justify the efforts I assume.

This issue may also arise with other tools when it comes to select big entities (circle, arc, ellipse) on high zoom level.

lucabalsanelli, 2022-03-01:
As long as the math is correct, that's fine.

jbergengh, 2022-03-01:
This issue does not occur on Qcad CE

[image]

lucabalsanelli, 2022-03-02:
My specific issue occurred even in QCad 3.27.1 CE

lucabalsanelli, 2022-03-06:
@lordofbikes should I close this issue?

lordofbikes, 2022-03-06:
We keep this open to keep in mind for V3.
Many thanks for asking.

pollywinters, 2022-03-13:
The problem does in fact seem to be that the on-screen rendering for circles draws them a very small fraction oversized. All the calculations for intersections with the circumference of a circle eg. snapping a line onto the edge of a circle, trimming a line that passes through a circle, seem to be correct with the generated straight line being exactly the right length and terminating at the expected coordinates. Needs a maths expert to look at whatever algorithm is used for the circle rendering.
Glancing through the code in file rs_painterqt.cpp, circles are treated as being the special case of an ellipse with major and minor axes the same length. Then in rs_painter.cpp ellipses are rendered using trigonometric calculations. Could be that a more accurate, and faster, rendering could be produced for circles using the Bressenham circle algorithm - draws circles using no trig or square root calculations, only multiplies and adds, I would have to dig out my old books on graphics algorithms though to find if the Bressenham method can be used for the case of general ellipses...