LibreCAD/LibreCAD #1538: Simple copy paste does not seem to work as expected

WolfgangFahl, 2022-04-16:
### Expected behavior
copy + paste should paste
### Observed behavior
Nothing happens
### Steps to reproduce or sample file
```
0,0
0,250
124,250
124,0
close
```
select - apply hatch (concrete)
[image]

select - copy - paste
### Operating System and LibreCAD version info
Mac OS 11.6.2
LibreCad : 2.2.0-rc3-23-g3c4163a9

WolfgangFahl, 2022-04-16:
Somehow the selection seems to be broken. 
Lets's start again and show the context.
```
0,0
600,0
600,600
384,600
384,480
264,480
264,360
close
```

WolfgangFahl, 2022-04-16:
[image]

WolfgangFahl, 2022-04-16:
```
pl
0,0
0,250
124,250
124,0
close
```

WolfgangFahl, 2022-04-16:
[image]

WolfgangFahl, 2022-04-16:
select and hatch
[image]

WolfgangFahl, 2022-04-16:
[image]

WolfgangFahl, 2022-04-16:
Now the hatched "tile" should be duplicated multiple times. Let's try this ... select 
[image]
[image]
copy 
[image]

paste
[image]

WolfgangFahl, 2022-04-16:
Instead of pasting the single tile the whole construct is pasted:
[image]

WolfgangFahl, 2022-04-16:
This is at least counter intuitive.

WolfgangFahl, 2022-04-16:
When removing the "unwanted parts" a second paste simply fails and does nothing.

WolfgangFahl, 2022-04-16:
cut is also unreliable and might or might not work - this is all very strange ...

lordofbikes, 2022-04-16:
Works for me as expected.
Presumably you missed the important step to set a reference point between `copy`/`CTRL-c` and `paste`/`CTRL-v`.
So the correct command sequence is `select` - `copy` - `reference` - `paste`.
See https://librecad.readthedocs.io/en/latest/ref/menu.html?edit#edit:

> A reference point needs to be placed for subsequent paste operations.

There is also the hint `Specify reference point` in the command line box and the status bar after `copy` command was executed.

Once copied or cut, with reference point set, multiple pastes were no problem for me.

WolfgangFahl, 2022-04-17:
This indeed seems the point. I suggest to change the unexpected behavior  to ask for a reference point and implement traditional intuitive cut&paste instead. The whole show would be made much simpler if it it a 100% what is selected at all times and the select object can be manipulated e.g. moved around, rotated and so on intuitively as is IMHO the whole idea of using a mouse. See e.g. the mother of all demos https://www.youtube.com/watch?v=yJDv-zdhzMY

sand1024, 2024-09-19:
I think copy issues (with and without reference point) this is addressed by adding "Copy Quick/Cut Quick" commands added to 2.2.2, most probably the issue may be closed