LibreCAD/LibreCAD #1543: Application paths are empty by default, and translations are missing on Fedora

martonlente, 2022-05-21:
Hello,

LibreCAD is available on Fedora 36 in Gnome Software, from Fedora Linux (RPM) repository. After installing LibreCAD, the **Paths are empty** though in Application Preferences dialog. This results primarily in the following issues:
- Translations are missing by default,
- Hatch Patterns are missing by default,
- Fonts are missing by default,
- Part Libraries are missing by default, etc.

The according packages get installed behind the scenes, and most of the above issues can be resolved by manually setting these:
- Hatch Patterns: `/usr/share/librecad/patterns/`
- Fonts: `/usr/share/librecad/fonts/`
- Part libraries: `/usr/share/librecad/library/`

However, manually setting the Translations' path to the installed `/usr/share/librecad/qm/` directory doesn't make translations available, **languages are still missing from Language settings.**

Is it maybe because that the RPM package is slightly broken? Is there a way to make translations available (at least manually, until setup issues are getting sorted out)?

Thank you in advance!

lordofbikes, 2022-05-21:
That the paths in Application Preferences are empty is correct, LibreCAD searches on standard locations for those files. Paths settings are for additional or non standard locations.

There were some modifications related with AppImage lately, maybe they broke something.
I'll check this.

martonlente, 2022-05-22:
Hello,

Thanks @lordofbikes for the quick reply, and clarification on the default (empty) state of paths in Application Preferences.

Let me add a quick summary, reflecting the above:
- The more precise definition of the error is, that **LibreCAD doesn't find standard locations** (e.g. `/usr/share/librecad/patterns`) by default on Fedora 36.
- The workaround to manually setup paths to correctly installed default resources work for hatch patterns, fonts, and part libraries (see above), but **doesn't work for translations.** LibreCAD still doesn't list languages in Language settings, even if translations' directory is correctly referenced in Application Preferences.
- The LibreCAD instance I'm referencing doesn't run as an AppImage, but from a standard Fedora Linux (RPM) installation. Is LibreCAD also available as an AppImage, and if so, should I try running that one?

Thanks!

lordofbikes, 2022-05-22:
I just filed a Fedora bug report: https://bugzilla.redhat.com/show_bug.cgi?id=2089001

This is caused by issue #1488, which is solved in our code base, but not in Fedora yet.

Many thanks for your efforts reporting this such detailed. :+1:

lordofbikes, 2022-05-22:
follow up: https://forum.librecad.org/LibreCAD-2-2-0-all-text-has-gone-no-fonts-tp5721474p5721701.html

martonlente, 2022-05-22:
Hello,

Thank @lordofbikes so much for the update, and managing this! :+1: I'll look forward to the Fedora package's update then.