LibreCAD/LibreCAD #1545: Various rotated entities shift from degrees to radians on file reload

DeliveryFry, 2022-06-04:
Hi everyone. Long time LibreCAD user, but first time bug reporter, here.

My Linux distro recently updated their packages from LC version 2.1.3 to **2.2.0rc3** (due to plugging a security hole in **libdxfrw 1.0.1**, _see_ [here](https://bugs.mageia.org/show_bug.cgi?id=29720)) and I've spent the last few weeks putting together a short list of both the new and longstanding bugs/issues I've found that annoy me the most.

The _most_ annoying of these is new (to me) starting with 2.2.0rc3 and what I've stated in the title: upon file reload certain types of rotated entities have their degrees shifted to radians such that if one is saved at 90°, it reloads as if it were now at 1.5708°. Saving it again reloads it at 0.0274156°, _etc._ until repeated saves dial these entities completely down to 0°.

Entities subject to this bug I've observed so far are **single-line text**, **hatch patterns**, and **dimension arrows/text**. Other entities, such as rotated blocks and multi-line text are _not_ affected.

From digging around for previously reported issues of this nature, what I'm describing bears suspicious similarities to the ages-old bugs #271 and  #707.

My Linux distro is **Mageia 8** (x86_64) and I'm using **Plasma 5** (KDE Frameworks 5.76.0
with Qt 5.15.2) as my desktop.

DeliveryFry, 2022-06-07:
> Can you share a DXF file for a check ?

Sure thing. But in case (for whatever frustrating reasons) you, or others, can't verify my issue, I'll do better than that.

Here's what the supplied drawing _should_ look like:
[image]
And this is how it looks after the first save, close, and reload cycle:
[image]

Using the `Properties Tool` on the hatch pattern, for example, reveals its freshly borked rotation angle (it should be 90°):

[image]

Here are **3 copies** of the DXF including the initial (first) save plus a second and third save iteration after things have started to go wonky. I did _nothing_ to the drawing between saves except `zoomauto`.

[LC-Bug-1545_Rotation-Issue.zip](https://github.com/LibreCAD/LibreCAD/files/8848892/LC-Bug-1545_Rotation-Issue.zip)

Thanks for taking a look into this matter; I greatly appreciate it.

lordofbikes, 2022-06-07:
Many thanks for the sample files.
Between 2.1.3 and 2.2.0-rc3 we have nearly 600 commits. I think I remember an issue in LibreCAD filter or `libdxfrw` related with wrong angle interpretation. Have to sort this out first.
Do you have by chance the original file saved in 2.1.3?

Just tested on Windows I can't reproduce this issue. So this is possibly caused by your distribution maintainer. I can imagine that mixing incompatible versions of LibreCAD and libdxfrw can produce such results. I'll try to check what they have used for their update.

You can try to use the [AppImage](https://github.com/LibreCAD/LibreCAD/releases/tag/latest) until we have sorted this out.

DeliveryFry, 2022-06-07:
Hi. Thanks for your reply.

> Do you have by chance the original file saved in 2.1.3?

Nope. I cobbled that example drawing together using my distro's **rc3** at most 20 minutes before posting the link to its zip in that last comment. Not that it likely matters because…

> So this is possibly caused by your distribution maintainer. I can imagine that mixing incompatible versions of LibreCAD and libdxfrw can produce such results.

Your guess is probably spot on. I gave the AppImage you recommended a quick go and can happily confirm that it does _**NOT**_ exhibit this bug. Yay!!!

> I'll try to check what they have used for their update.

I think I can help with that. Mageia is an RPM distro. Here are links to our App Database's source packages for the latest builds we're using: [LibreCAD](http://madb.mageia.org/rpm/show/application/0/release/8/arch/x86_64/name/librecad-2.2.0-0.rc3.1.mga8.src.rpm/source/1/t_media/4); [libdxfrw](http://madb.mageia.org/rpm/show/application/0/release/8/arch/x86_64/name/libdxfrw-1.0.1-1.1.mga8.src.rpm/source/1/t_media/4). From there, Sophie links to the files' contents appear at the bottom of each page.

I'll spend the next day or two playing around with this latest AppImage version to see if there are any more bugs I can cross off my "List of Most Annoying Annoyances" without ever needing to report about them.

Thanks muchly for your help thus far.

lordofbikes, 2022-06-07:
I just filed a bug report on Mageia:
https://bugs.mageia.org/show_bug.cgi?id=30521

They added unmerged LibreCAD/libdxfrw#51 pull request, which cause this issue.

DeliveryFry, 2022-06-07:
> I just filed a bug report on Mageia

Super! I've already replied and thanked you over there, but really can't thank you enough, especially for the prompt analysis.

DeliveryFry, 2022-06-09:
My distro's update candidate for the new **2.2.0rc4** looks good. I'm happy with it, at any rate. So, we can scratch this bug off our lists as fixed?

lordofbikes, 2022-06-09:
Glad to read that :+1:

closed 2022-06-09 by lordofbikes