LibreCAD/LibreCAD #1548: v2.2.0-rc4 M1 Build Doesn't Open

jabelone, 2022-06-11:
Hey there,

I have an issue running the v2.2.0-rc4 M1 build on a MacBook Air M1 running macOS Monterey 12.4. My understanding from the release notes is that this is the first time an M1 build was attempted, so I thought I'd report that it unfortunately does not run. Note however, that the x86 version from the same release opens and appears to run via the Rosetta 2 translation layer.

Also something else I thought I'd mention is that the file is called AMD64, but typically, M1 native builds are referred to as ARM64 or just Apple Silicon (at least in my experience).

### Expected behavior
LibreCAD launches without an error.

### Observed behavior
The application throws an error saying it is damaged (as per screenshot below):
[image]

### Steps to reproduce or sample file
I have just tried to download the M1 build (`LibreCAD-2.2.0-rc4-amd64.dmg`) from the v2.2.0-rc4 release [here](https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.0-rc4). I open the .dmg, drag the `LibreCAD.app` file into either Downloads or Applications. I then attempt to open it by right clicking then clicking open (as it's an unsigned package, the right click open step is required). At this point, macOS displays an error saying the application is damaged (as per screenshot above).

### Operating System and LibreCAD version info
LibreCAD: v2.2.0-rc4
System: MacBook Air M1
Operating System: macOS Monterey 12.4

lordofbikes, 2022-06-13:
Many thanks @jabelone for reporting.
Yes, it was a try, I'll remove the package from downloads to avoid any further issues.
Good to hear, that the x86 package works at least.

Sorry for the irritation about the architecture label, my fault, of course it is arm64.
Or was it auto-correction :thinking: :rofl:

I'll keep this open, in case we start another try.

jabelone, 2022-06-14:
No problem, all the naming conventions are a little confusing tbh. 😅

Unfortunately that build appears to have the same issue as the original one.

lordofbikes, 2022-06-14:
Many thanks for testing @jabelone! :+1:

qcksilver, 2022-07-04:
Do you need more testers for a native M1 version? 
I also have a M1 14" MBP (2021) and I can test if needed.

gnattu, 2022-10-24:
I managed to build it from source and run it as arm64 native on macOS 12.

I think the "app is damaged" dialog is indeed caused by the binary not been correctly signed, and option+open is not enough on Apple Silicon. 

For Apple Silicon to run unsigned binaries downloaded from internet, you need to use the following in console: 
```shell
xattr -dr com.apple.quarantine /path/to/application.app
```
This will remove the quanrantine xattr, so that macOS will sign the binary ad-hoc on first launch and then get trusted by the security mechanism of Apple Silicon Macs(mandatory signed binaries on macOS). I cannot test if this is the really the case because the download link to the built binary seems to be removed, but I highly doubt it is as the master branch build and runs on my Macbook Pro M1 Max.

How I build LibreCAD:
-----

- git clone the master branch
- `brew install qt@5 boost `
- `/opt/homebrew/Cellar/qt@5/5.15.6/bin/qmake librecad.pro -r -spec macx-xcode` This requries you have Command Line Develop Tools from Apple, and you have to use full path of qt@5 because default qmake is for qt6, and this project does not build against qt6
- `make -j4` as the doc says, probably can be built on `-j10` as well.

Then I can launch the LibreCAD.app which is native arm64:

`file /Users/gnattu/src/LibreCAD/LibreCAD.app/Contents/MacOS/LibreCAD
/Users/gnattu/src/LibreCAD/LibreCAD.app/Contents/MacOS/LibreCAD: Mach-O 64-bit executable arm64`

[image]

[image]

I have not test all functionalities intensively but at least it runs.

CodiePP, 2022-10-30:
I was using the following command to setup & compile on Apple M1: 
```
/opt/homebrew/Cellar/qt@5/5.15.6/bin/qmake librecad.pro -r -spec macx-clang-arm64 BOOST_DIR=/opt/homebrew/Cellar/boost/1.80.0 CONFIG+=sdk_no_version_check
gmake -j10
```
the resulting app:
```
% file ./LibreCAD.app/Contents/MacOS/LibreCAD
./LibreCAD.app/Contents/MacOS/LibreCAD: Mach-O 64-bit executable arm64
```
works nicely.

lordofbikes, 2022-10-30:
This issue is not about building and running LibreCAD on the machine it was built.
We need a valid composed `DMG` package which will run on other `arm64` hardware without build environment. This failed so far.

There is `scripts/build-dmg.sh` which works on non `arm64` hardware and creates the `dmg` package. This script needs probably be adapted for `arm64` or possibly `scripts/postprocess-osx.sh` is the problem.

gnattu, 2022-10-30:
`This failed so far.`

As I stated, this is probably not the build is failed, it is only that the binary is not properly signed and is rejected by Apple Silicon's security policy. (macOS will tell the user the app is damaged when it is not properly signed on Apple Silicon Macs).

And I provided a potential workaround to use `xattr -dr com.apple.quarantine /path/to/application.app` to remove the `com.apple.quarantine` flag so that macOS will sign the binary ad-hoc on first run instead of refusing to run at all, but I cannot verify that because the download link to the binary file in question is removed, so that I have to build it from source to check if it's LibreCAD really does not run, and that is not the case.

lordofbikes, 2022-10-30:
The `dmg` package for Intel hardware is also not signed and right, there is some kind of warning the user must accept to run the unsigned package/binary.
Because the reported messages are different, I assume a problem with packaging.
There are probably libraries missing, which must be present on `arm64` but not bundled.

You can download the package in question here:
http://download.librecad.de/LibreCAD-2.2.0-rc4-arm64.dmg

But for a valid test, it should be run on a system without build environment.
Otherwise probably missing libraries may be found in system paths.

gnattu, 2022-10-31:
I've tried with that binary and it successfully run on two of my Macs, one has the build environment, the other does not.

Before I did `xattr -dr com.apple.quarantine /path/to/application.app`, I received exactly the same error as expected on any unsigned binary:

[image]

But after `xattr -dr com.apple.quarantine /path/to/application.app`, it launches just fine:

[image]

Such workaround is mandatory now on any Apple Silicon Macs to run unsigned binaries downloaded from internet, probably we can include a txt file inside the dmg to notify users of such change.

JeroenGar, 2022-10-31:
I can also confirm this works on my 14 inch MBP

niXta1, 2022-11-16:
There is no problem with the build for arm64, it't just not signed. Unsigned binaries will output the error, damaged.

Put an exception on it or sign it, those are the options. Many open-source apps have the same situation.

Just put this in the instructions and you're done:
`xattr -dr com.apple.quarantine /path/to/application.app`

jabelone, 2022-11-17:
@lordofbikes sorry I've been super busy at work! I forgot to mention that I can confirm it's because the application binary isn't signed (correctly). Apple has been making it harder for incorrectly signed apps to run on macOS. This is generally a good thing :tm: as it makes malware and bad actors much easier to track and block. However, it puts a higher burden on developers to maintain signed packages.

You can see below with [this signing utility](https://objective-see.org/products/whatsyoursign.html) I've been using to troubleshoot an installer package at work. It shows the binary isn't signed at all, which is why macOS considers it "damaged". The correct way of fixing this is to sign it with a valid Developer ID Application Certificate (requires a paid Apple Developer account) and then notarise the application with the Apple notarisation service as per the instructions [here](https://stackoverflow.com/questions/69354021/how-do-i-go-about-code-signing-a-macos-application). You could use a tool like [Fastlane](https://docs.fastlane.tools/codesigning/getting-started/) which makes the process a bit easier (especially the notarisation part).

[image]

p.s. ordinarily I'd offer to help sort this out, but I'm very time poor right now and won't want to commit to something I may not be able to follow through on. Hopefully those links will get you started in the right direction if you want to sign the macOS builds correctly.

jabelone, 2022-11-17:
Using the workaround by @niXta1 and others above works fine with the "damaged" build. However, this is not an acceptable solution unless you're a developer. If the maintainer(s) don't have the resources to maintain a signed build, it's reasonable to not offer one. However I'd strongly discourage instructing non-devs to bypass package signing requirements. Gatekeeper and package signing is specifically designed to stop malware and known bad actors etc. If a "legitimate" application instructs a user to run unsigned packages, they may be more likely to do so with malware etc.

lordofbikes, 2022-12-18:
Just a short notice, that I did not forget about this, focus was on the 2.2.0 release recently.

I still can't get my head around it, that Microsoft and Apple rise the requirements for software with each system evolution. The issues on Windows, with unsigned software, becomes more and more worse too.
It's not about signing, I understand the security idea behind it. It's more about the fact, that free open source communities are forced to pay regularly for the certificates. This does not complain with my imagination of the free open source idea.
And there are still no efforts from those big players, which would enable open source projects to apply for free certificates. If it's just me, who missed it, please give me a hint.

Anyhow, I'll use the holiday season to think about if we should apply for the needed code-signing certificates.
But the problem is not just paying for the certificates, they have to be used appropriate. That means we need to get the skills and implement it in CI, what consumes valuable resources.
Especially for the Apple-ARM platform, there is also the need for an appropriate system to build the package. This is still in beta testing for GitHub Actions, what would be the best option, as it fits into the current build systems.
Renting ARM hardware in the cloud is not only the costs, it also needs to implement an appropriate automatic build system there.
Concerning the costs, the rent for about a year is as much as the price for buying our own hardware.

So probably a few things to think about during holidays.

ashish3805, 2026-05-03:
### **Fix for "App is Damaged" / Crash on Apple Silicon (M1/M2/M3)**

If you are on an M-series Mac and the app fails to open or crashes immediately with a "Code Signature Invalid" error, follow these steps:

1. **Install Correctly**: 
   * Open the downloaded `.dmg`.
   * **Drag** the LibreCAD icon into your **Applications** folder. 
   * *Important: Do not try to run it directly from the DMG window.*

2. **Clear Quarantine**: 
   Open **Terminal** and run:
   ```bash
   xattr -cr /Applications/LibreCAD.app
   ```

3. **Self-Sign the App**: 
   In **Terminal**, run this command to authorize the app for your chip:
   ```bash
   sudo codesign --force --deep --sign - /Applications/LibreCAD.app
   ```
   *(Type your Mac password when prompted and hit Enter. You won't see characters as you type.)*

4. **Launch**: 
   **Right-click** LibreCAD in your Applications folder and select **Open**.

---

**Why this is needed:**
MacOS flags unsigned open-source binaries as "damaged." These commands manually strip the warning and "ad-hoc" sign the code so the ARM64 architecture allows the process to run.