LibreCAD/LibreCAD #1553: Import error message when starting LibreCAD

lucasgamaleri, 2022-07-06:
### Expected behavior
Expected LibreCAD to open into a new file when starting.

### Observed behavior
When you start LibreCAD, it shows an Import error: error opening DXF/DWG file.

### Steps to reproduce or sample file
Open LibreCAD in default configuration, in my case I just installed from Sourceforge.
It's the first error message that appears

### Operating System and LibreCAD version info
Operating System:
Edition Windows 10 Home
Version	21H2
OS Compilation 19044.1806

LibreCAD version info:
Version: 2.2.0-rc3-24-g4602ec6a
Compiler: GNU GCC 7.3.0
Compiled on: Mar  5 2022
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)
[image]

lordofbikes, 2022-07-06:
Works for me!
Can you detail your steps?
Any issues during installation?
Used administrator privileges anywhere?

lordofbikes, 2022-07-24:
Had this recently too, but no chance to investigate.
But pretty sure, this is a Windows only issue.

lucasgamaleri, 2022-07-24:
I agree it must be a OS specific issue. I couldn't build the project from source on VS nor Qt by following the repository wiki so couldn't investigate. I might keep trying in the future

closed 2026-09-26 by sand1024