LibreCAD/LibreCAD #1555: Trimming circles with line

Vilcrow, 2022-07-24:
This is my first experience of participating in such a project.
### Expected behavior
Cut the length of a circle entity to an intersecting line entity.
### Observed behavior
Sometimes the circle isn't trim off (it seems only visually).
### Steps to reproduce or sample file
1. Draw line (doesn't matter how).
2. Draw circle (doesn't mater how) which intersects our line from the first step.
3. Activate Trim tool.
4. Select the line as limiting entity.
5. Select the circle as entity to trim.
6. Show result.
7. If everything went well repeat steps 2-7.
### Operating System and LibreCAD version info
Version: 2.2.0-rc4-8-g5a07ab41
Compiler: GNU GCC 9.4.0
Compiled on: Jul 23 2022
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.2 LTS

https://user-images.githubusercontent.com/96838551/180628461-cc1f305c-247f-47bd-9654-c17c847cb017.mp4

https://user-images.githubusercontent.com/96838551/180628462-613b343d-5468-4689-9b9a-b268b7245fd5.mp4

https://user-images.githubusercontent.com/96838551/180628465-30473602-55b8-420b-8647-e50791577318.mp4

https://user-images.githubusercontent.com/96838551/180628469-e6b26975-96f7-44a8-896f-08ccd4645b1b.mp4

https://user-images.githubusercontent.com/96838551/180628472-301eabdb-cf53-4329-adc6-5b64b67ca04b.mp4

lordofbikes, 2022-07-24:
Cross link, as this is similar to #818

First, I agree, that trimming with circles has some room for improvement.

But, there are so many possibilities and different circumstances, that it is tough to pack them all into one tool.
So the current implementation has some logic which works for some cases, but also fails on others.

Most important, as I have seen in most videos above, is to understand the difference between CIRCLE and ARC entities. For CAD these entities are different in their definition and handling.
Generally, when a CIRCLE is trimmed, it becomes an ARC.
When a CIRCLE only needs center and radius to describe the geometry, an ARC has also a start and end angle.

So what happens when there is no visual reaction after trim, the CIRCLE was converted to an ARC, but with identical start/end angle. Thus the CIRCLE has become a 360° ARC.
When the first trim is applied by the rules of CIRCLE, following trims will be handled by ARC rules.
To identify the entity type it has to be selected, because a not selected CIRCLE and 360° ARC are identical.
But when they are selected, there is one little difference. Both entities have a blue square in their center, but only the ARC has another blue square on the circumference. This is where the CIRCLE was trimmed and shows the start/end angle of the arc.
This is what happens in most videos above, first the CIRCLE becomes an ARC and on second trim one of the start/end angles is trimmed, when the other stays where it was trimmed on the CIRCLE.

As said in the beginning, it is hard to estimate the users desire with one click.
It is regularly topic in the forum, but there was no solution so far which would not have created other issues or break current behavior.

I avoid to trim circles when I'm not sure, from my experience, that it will work.
Usually, in a progressed drawing, there are already intersections where a circle should be trimmed. So I divide the circle twice and delete the unneeded part. This is not much more work, but avoids trouble with unexpected trim results.

Vilcrow, 2022-07-24:
Thanks @lordofbikes for the quick reply.
I figured out the logic of this operation. It works. 😁
I didn't pay attention to the difference between CIRCLE and ARC.

closed 2022-10-30 by lordofbikes