LibreCAD/LibreCAD #1557: When close LC, it opens a drawing windows

tochimoreno, 2022-08-12:
### Expected behavior
* Close LC
* No more LC windows

### Observed behavior
* Close LC
* A full-screen windows, only the drawing, not menus nor toolbar, appears for every drawing in LC
* Need to close each one individually and accept to save o not

### Steps to reproduce or sample file
* Open LC
* Open a drawing.dxf
* Close LC
* Close drawing.dxf
* Chose save o not
* Close unnamed document 1
* Chose save o not
...

### Operating System and LibreCAD version info
Version: 2.0.7-1877-gd702618c
Compiler: GNU GCC 11.2.0
Compiled on: Jun 17 2022
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Ubuntu 22.04 LTS

* Drawings:
  * Windows mode
  * Arrenge: Maximized
  *

lordofbikes, 2022-08-12:
> Close LC

How? File menu? Keyboard short-cut? Window-Icon?
Can't reproduce it, works for me.

tochimoreno, 2022-08-12:
Windows icon.

I'm using LXQT.
Te versión I use is the last. I download from here and compile in muy PC.

lordofbikes, 2022-09-07:
Found the time to check this and my opinion is, that this is intended behavior.
Not from LibreCAD, but from Linux or probably other OS too.

When LibreCAD is not in full-screen mode, you see two close buttons. One in the colored window title bar and one inside the window.
The title bar icon is to close the application and the inner icon is to close only the active document.
So in full-screen you only see the inner icon to close the document, the title bar and the application close icon are out of screen.

To close the application in full-screen you can use `CTRL-q`, `ALT-F4` or the file menu. Then, in case of unsaved changes, you can save or close all at once.
Or switch full-screen off (`F11`) to make the title bar and application close icon visible again. Then you can use the title bar icon to close the application and all documents.

closed 2022-09-07 by lordofbikes