LibreCAD/LibreCAD #1558: Elements of exploded dimensions do not shift to a different Layer.

ajk2196, 2022-08-23:
The exploded elements of a dimension do not shift to a different layer, neither by first selecting the element and double clicking on the layer, nor by first selecting the element and then modifying the attributes to make them belong to a different layer.

Expected behaviour: The exploded elements of a dimension are lines and a number in text format. These should easily switch layers when made to switch them.

Observed behavior: The elements independently never change their layers.

Steps to reproduce: Firstly add two new layers with different line colours. (Go for green and blue colour for easiness). Add a line. Add a dimension to that line. Select the dimension. Explode it. Try changing the layers of the exploded dimension to a different layer. That won't happen.

Operating System and LibreCAD version info:

Version: 2.2.0-rc4-8-g5a07ab41
Compiler: GNU GCC 7.3.0
Compiled on: Jul  3 2022
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

dellus, 2022-08-23:
For me exploded elements of a dimension do shift to a different layer. They just don't take the attributes of the new layer. Out of the box  the attributes settings for dimension elements are "By Block". It seems this is kept when exploding. Change the attributes/properties to "By Layer" after exploding and all is well.

ajk2196, 2022-08-23:
Ok, seems to work that way. But I think that this is a confusing behavior. It should by default change the layer along with all the attributes by default - just like all other lines and texts do.

lordofbikes, 2022-09-07:
Just for curiosity, what is the reason for exploding dimensions?
Is this a frequent task or just needed exceptionally?

ajk2196, 2022-09-09:
It has been long now. I think that back then, I needed it to make the dimension lines and fonts in different colours - wanted the dimension lines to be dark blue and dimension in slightly lighter blue. I wanted it badly back then.

ajk2196, 2022-09-10:
@lordofbikes , now after reopening my project, I remember why I needed to explode the dimensions. Please have a look at this video. The main problem here is that the text still has a black outline, while it should have changed to a dark green one. Also, the exploded text doesn't change layers upon firstly selecting them and then double clicking on the layer name.

Also, this is frequent because I do not like a hollow text. Single line texts barely have any thickness in printed large sized plan (Say on A1 sized page) - which makes them almost invisible even to a good healthy eye. Plus, as I have said before and am reiterating here again, we need better fonts with smoother curves and corners to give a professional look to the drawings.

https://user-images.githubusercontent.com/90527174/189492720-e33eb8ce-5fa9-4538-a632-ed1458316ead.mp4

dellus, 2022-09-10:
A bit off topic, but interesting you have found a way to make non sticky, filled text like with ttf. I have tried myself too before and found it didn't work satisfyingly. Now as you had success I tried it with the komimincho font you used in the video. And indeed, it works! But not with all hollow fonts. Greek ol and amiri regular (nearly identical with komimincho) work ok too. In some zoom settings you get artifacts, sometimes also in Print Preview. But when I exported to pdf they were gone.
Wish there was a hollow font more similar to something like Arial. I tried to make it with ttf2lff, but failed. Someone else?
For dimensions standard sticky font is ok for me, but for titles of drawing parts a fat ttf like text would be nice.