LibreCAD/LibreCAD #1560: Offset and/or Parallel feature is not working.

ajk2196, 2022-08-29:
Offset and/or Parallel feature is not working.

Version: 2.2.0-rc4-8-g5a07ab41
Compiler: GNU GCC 7.3.0
Compiled on: Jul  3 2022
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

ajk2196, 2022-08-29:
Please check this video for the issue. Not only this, I cannot even select a line like I used to in previous versions - by doing a single click.

https://user-images.githubusercontent.com/90527174/187181512-a7b0e0e1-aec7-4d3d-a29d-8755674bee01.mp4

ajk2196, 2022-08-29:
Seems like I have figures out what the problem is. This error happens when working with large numbers of dimensional units. In my actual drawing, I was using millimeters and was working with a rectangle of 3500mm x 7500mm when I cam across this problem. After posting this issue, I tried scaling down the drawing and everything worked fine like before. I could even select lines by clicking on them. The problem seems to be with something similar to 'Pick Radius' of Auto CAD. In LibreCAD, the pick radius does not seem to be screenspace related, i.e. it does not effect or consider something when it is 'x' pixels away from the cursor, but only effects the things that are 'x' millimeters away from the cursor. There should be an option to select the pick radius. Atleast it should be mad screen-space adaptive.

fa201, 2022-08-29:
I have noticed such a behavior but not to this extent. My work-around is _zoom in_, _select the feature_ and _zoom out_. I agree this this a bit annoying.

@lordofbikes @feragon Can we get you opinion on this for LibreCAD V2 and V3 ?

ajk2196, 2022-08-31:
Yeah. The real problem is when working with millimeters. A room can easily be 4500mm x 6000mm or more. Now in my plan, I had around 8 rooms around the above mentioned size. Now to see what lines are being offset in the entire plan, I have to zoom out enough to adjust all those eight rooms on my screen. And when I try to offset with zoomed out, it never happens. Like @fa201 said, we have to zoom in for each corner to offset two separate lines, and this is tiring and also annoying. **I know only little bit of coding, which is why I am suggesting this next advice:**

Logically the code for offset would be something like this:

>Offset if:
>   Cursor within x distance from the line to be offset.

This should be changed to:
> Offset if:
>   Cursor within (x distance / Zoom factor) from the line to be offset. #This is considering that the zoom factor decreases when the drawing is zoomed out. If Zoom factor increases while zooming out, then multiplication should be used instead of division.

lordofbikes, 2022-09-07:
I just checked this on Linux and Windows 10.
Can't reproduce this issue, even with 100000 long line I can easy use parallel tool and select the lines with single click.

Maybe a test case DXF file can help to reproduce it.

ajk2196, 2022-09-10:
I tried same with a 100000 long line. Here is my screen-record. Sometimes things work, most of the times they don't - there is no consistency and reliability. Plus the workaround suggested by @fa201 is appreciated, but if you see it from my perspective, it is not affordable in large scale high detailed drawings where similar long lines are just too many and are too close to each other. The offset tool in such cases mostly never shows the dotted offset projection. I wish I could have shown you a demonstration but my work is collaborative and I alone don't have any right to publish anything from it anywhere. 

Also, in this case, copying/pasting a line at a distance with ortho mode ON can be a good workaround, but as you may see in the video itself, selecting longer line with a zoomed-out view too is tough - you have the click at the exact middle on the line. Note that this workaround of copy/pasting too is a lot of hassle. We need to mend the offset tool and prevent adopting workarounds.

https://user-images.githubusercontent.com/90527174/189492027-f170fcce-a189-4ff0-a507-3964d3e0e431.mp4

fa201, 2022-09-10:
I think it is related with the snap/selection feature: the "search radius" seems to be in real units (those specified as drawing units) where we expect it to be pixel units (this is zoom independent)