LibreCAD/LibreCAD #1561: Unable to insert nested blocks in Librecad 2.2.0

bean-gh, 2022-09-06:
### Expected behavior

Nested blocks should be created when inserting a block containing blocks.

### Observed behavior

The functionality was working in 2.1.3 version. Does not work on 2.2.0-rc3 or 2.2.0rc-4.

### Steps to reproduce or sample file

Attached block 0-001.dxf
[0-001.zip](https://github.com/LibreCAD/LibreCAD/files/9494696/0-001.zip)

### Operating System and LibreCAD version info
Working system :
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 
[image]

Broken System:
Version: 2.2.0-rc4
Compiler: GNU GCC 12.1.0
Compiled on: Jun  7 2022
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)
[image]

bean-gh, 2022-10-04:
I am wondering whether the "Insert" method discusssed here; https://ezdxf.readthedocs.io/en/stable/tutorials/blocks.html#block-references-insert , may be the solution for all of these issues:

#1499 opened on 8 Jan by [melwyncarlo]
#1344 opened on 2 May 2021 by [allanwmacdonald]
#1324 opened on 24 Feb 2021 by [matthijskooijman]
#1284 opened on 29 Sep 2020 by [tochimoreno]
#1223 opened on 18 May 2020 by [JohnCC330]
#1223 opened on 18 May 2020 by [JohnCC330]

All of the issues liisted above seem to be related to nesting (wrapping) of blocks.

dxli, 2023-06-06:
yes, we got some serious regression

a good fix is still needed

Could you test the quick fix in the latest master branch?