LibreCAD/LibreCAD #1563: Arch-specific libraries do not go in /usr/share

hobbes1069, 2022-09-14:
Arch specific files (shared libraries in this case) do not get installed into `/usr/share` which is for non-arch data only. This is per the FHS specification:

https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard#Directory_structure

For 32-bit libraries (nearly obsolete in Fedora) the proper location would be `/usr/lib`, but for a 64 bit architecture, the correct location is `/usr/lib64`

https://github.com/LibreCAD/LibreCAD/blob/5a07ab418ab23f2f2553d267eeb15a09c1220ec0/librecad/src/lib/engine/rs_system.cpp#L591

Also, to be clear, it needs to be one directory or the other, adding both to the search path could cause attempts to load a 32bit plugin on a 64 bit system (however unlikely).

hobbes1069, 2022-09-16:
How hard would it be to add a qmake option for `PLUGIN_DIR`?

lordofbikes, 2022-09-17:
What is the intention of this issue?
Are you addressing an actual problem?
Are you a distribution maintainer having issues with this?

hobbes1069, 2022-09-17:
> What is the intention of this issue?

To get it fixed?

> Are you addressing an actual problem? 

Yes, the plugins are installed correctly into /usr/lib{,64} but the librecad binary does not look for them there, so they are unavailable.

>Are you a distribution maintainer having issues with this?

Yes, for Fedora Linux.

lordofbikes, 2022-09-17:
Thanks for the update.
Referencing https://bugzilla.redhat.com/show_bug.cgi?id=2126705 in the issue description would have saved me some hours :cry:

We also search in `/usr/lib` where the plugins were still found until now.
https://github.com/LibreCAD/LibreCAD/blob/5a07ab418ab23f2f2553d267eeb15a09c1220ec0/librecad/src/lib/engine/rs_system.cpp#L590-L593
Ignore the comments, they don't matter and I can't say how old they are.
When Redhat switched from `/usr/share` to `/usr/lib` in past, files were found anyway by the Debian style.

So new is, that you put the plugins to `/usr/lib64` now, which is not on our search list yet.
Obviously they were in `/usr/lib` in the Fedora 2.2.0-rc3 package, otherwise this issue would have been emerged earlier.

My interpretation of the FHS Wikipedia reference is, that `/usr/lib64` is optional.
Standard is `/usr/lib` for the host architecture and `/usr/lib<qual>` is the alternative for divergent architectures. That's what I know from various distros, `/usr/lib` and `/usr/lib32` for 32-bit files on 64-bit systems.

Anyhow, I'm not firm with distribution packaging. When Fedora puts the plugins in `/usr/lib64` now, we have to adopt this somehow.
Is there any reference from Fedora or Redhat addressing this? Because of:
>Also, to be clear, it needs to be one directory or the other, adding both to the search path could cause attempts to load a 32bit plugin on a 64 bit system (however unlikely).

We need a clear definition to find the best solution. How can we detect one or the other? Is there a date or Fedora release number when this changed?

Just for reference:

This seems to be related with a recent forum post or at least addresses the same issue:
https://forum.librecad.org/Dimensions-have-changed-inconsistently-tp5718417p5722066.html
I didn't realize the `lib64` in the path, just remembered #1488.
#1488 was about missing fonts, which was caused by a bug related with AppImage changes and fixed with 2.2.0-rc4.

There was also #1545, which was reported from Mageia users, a Fedora branch. It was caused by applying an erroneous, unmerged pull request for `libdxfrw`.
https://bugs.mageia.org/show_bug.cgi?id=30521

hobbes1069, 2022-09-17:
I'm not the primary maintainer of LibreCAD on Fedora so I can't say how (or if) it worked before, but I've been using Fedora since Fedora Core 10. It's always had the /usr/lib, /usr/lib64 to support multiarch, specifically to run 32bit binaries on 64bit systems. Debian has an arch tuple but I assume they have another way of finding libraries because there's too many to hardcode.

The choice to use /usr/lib64 is a distro choice, not an application choice :)

Interestingly we have RPM macros to enforce a lot of packaging standards including build flag, but also install locations. The qmake macro does not have any of this. It must not be standardized?

The actual output is quite long so I'll trim it down to what we care about for this issue.

```
$ rpm -E %cmake
<snip>
  /usr/bin/cmake \
        -S "." \
        -B "redhat-linux-build" \
        -DCMAKE_C_FLAGS_RELEASE:STRING="-DNDEBUG" \
        -DCMAKE_CXX_FLAGS_RELEASE:STRING="-DNDEBUG" \
        -DCMAKE_Fortran_FLAGS_RELEASE:STRING="-DNDEBUG" \
        -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON \
        -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF \
        -DCMAKE_INSTALL_PREFIX:PATH=/usr \
        -DINCLUDE_INSTALL_DIR:PATH=/usr/include \
        -DLIB_INSTALL_DIR:PATH=/usr/lib64 \
        -DSYSCONF_INSTALL_DIR:PATH=/etc \
        -DSHARE_INSTALL_PREFIX:PATH=/usr/share \
%if "lib64" == "lib64" 
        -DLIB_SUFFIX=64 \
%endif 
        -DBUILD_SHARED_LIBS:BOOL=ON
```

Or bog standard autotools configure:
```
$ rpm -E %configure
<snip>
   ./configure --build=x86_64-redhat-linux-gnu --host=x86_64-redhat-linux-gnu \
	--program-prefix= \
	--disable-dependency-tracking \
	 \
	--prefix=/usr \
	--exec-prefix=/usr \
	--bindir=/usr/bin \
	--sbindir=/usr/sbin \
	--sysconfdir=/etc \
	--datadir=/usr/share \
	--includedir=/usr/include \
	--libdir=/usr/lib64 \
	--libexecdir=/usr/libexec \
	--localstatedir=/var \
	--sharedstatedir=/var/lib \
	--mandir=/usr/share/man \
	--infodir=/usr/share/info
```

So I think there are a couple of options.
1. Assuming we stop looking once we find them, searching /usr/lib64 first would be fine because it will never be present on a 32bit system.
2. The other option would be to create a qmake build variable that allows it to be built into either a generated header or source file.

hobbes1069, 2022-09-17:
Question, where are the per user settings saved on linux? I need to be able to get to the initial settings. The 2nd and 3rd option are blank on Fedora. I believe the 2nd is language/translation location related.

lordofbikes, 2022-09-17:
>The choice to use /usr/lib64 is a distro choice, not an application choice :)

No one doubted this :wink: 

All current Fedora releases use `/usr/lib64`, I even found an old VM in my archive, with Fedora 30, which has `/usr/lib64` and non working LibreCAD plugins.
So this is not new, but obviously not detected for years.

I'll fix this for the final 2.2.0 release.

>where are the per user settings saved on linux?

`/home/{$USER}/.config/LibreCAD/LibreCAD.conf`

But what do you mean with *The 2nd and 3rd option*?
The `conf` file is created with the first start of LibreCAD, when the user is asked for his preferred language and the default measurement unit.

hobbes1069, 2022-09-18:
On initial startup without a configuration file there is a startup window with 3 setting dropdown boxes, one for units, one for language, and I can't remember the 3rd one. The latter two are blank/empty so it must not be finding the translation files at a minimum.

hobbes1069, 2022-09-18:
Ahh... just removed the config file and ran librecad. The GUI Language and Command Language...

lordofbikes, 2022-09-18:
I've seen this yet in my Fedora 30 VM. The language values in the `conf` was something like `$Invalid`. But it was able to select both languages in *Application Preferences*.
I think I also deleted the `conf` and had empty language boxes on the next start. But it was possible to select the languages in the start-up dialog. It seems to be an issue of initializing the boxes with a valid index, the language files were found. I'll check this too.

closed 2022-09-24 by lordofbikes

lordofbikes, 2022-09-24:
I considered your suggestion about using QMake variables or preprocessor directives for an arch specific build process.
But there are so many options, differences and compiler to consider across all the Linux distributions and their multiple supported versions that I canceled this idea.
As you already stated, 32-bit is obsolete, not only on Fedora. And having 32-bit *and* 64-bit LibreCAD on a system is indeed unlikely too.
So I came up with this solution to simply add `/usr/lib64` beside `/usr/lib`.
It worked for me on current 64-bit Ubuntu and Fedora.

Concerning the languages, I tested too and found it working with 2.2.0-rc4.
So the missing settings in `librecad.conf` were caused by the known issue with 2.2.0-rc3.
Deleting `librecad.conf` is not needed on updated systems, language settings can be set in *Application Preferences* when languages are found.

hobbes1069, 2022-09-24:
Yes, I deleted it to re-simulate a new user experience to see if it was working. 

Everything seem to be fine now!

commit 6f74b427 (2022-09-24): add /usr/lib64 to search for plugins, fixes #1563