LibreCAD/LibreCAD #1567: Deleting entities triggers repeated flood of RS_Modification errors

DeliveryFry, 2022-10-17:
Hi everyone. Linux user.

For as long as I've been working with LibreCAD (which is ages and ages) I've noticed the occasional slight delay when removing entities. No big deal.

But recently I've been working on what for me is quite a large project. I'm not sure what the rest of you consider "large" but so far this DXF is more than 307k lines long and weighs in at over 2 MiB. On the old (2008) hardware I'm still using, this project takes quite a while (20\~30 seconds?) to load. All good, no worries so far.

Then, maybe because of the combination of my old hardware with the size of this project, I've *really* started to notice a **delay** when removing entities. Except the delay is never predictable or consistent. Sometimes removing several dozen entities takes a spit second but removing just *one* can take up to 3 (!) full seconds. I started to think this was weird. I can `move/copy` entire *layers* comprised of hundreds of entities nearly instantaneously so why does *removing* a handful take so long?

Then I began to notice that during the longest of these delays LC would stall or freeze. For example, I would select some entities, press `[DEL]`, my mouse cursor would grow a little red 'X', and while the remove action was in progress, spinning my mouse wheel to zoom in or out would do nothing. It's very similar to the (understandable) freeze that I experience whenever the auto-save kicks in.

Then I also noticed that every so often — but not always — this delay is accompanied by what to me looked like excessive HDD activity. By this I mean that my HDD's LED glows steadily for the exact duration of the removal action. WTF?

Investigating what LC could possibly be doing with my HDD during these long intervals eventually led me to my **journal log**. Oh boy!

There I discovered that **every time** I've been removing so much as a single entity LC has been flooding my journal *dozens*, *hundreds*, often *thousands* of times per second with the following error message:

```
RS_Modification::remove: no valid container is selected
```

All I could gather from an online search for that message was that it comes from within the Docker API. Maybe this happens when some of the entities being removed are completely or partially outside the workspace window? That's just a wild guess. I really have no clue because I haven't found any way of reproducing the behavior reliably. That, and I'm more focused on getting my drawing finished.

So I think the best I can do for you folks is to share the relevant lines off one of my recent journal logs. Here's one from a couple of days ago. LibreCAD had `pid [18900]` so I `grep`'ed for that and then ran the result through `uniq -c` to get the counts on the repeats. That should give you a sense of how many times per second it hammers out the error. Contiguous seconds should constitute a single remove action. I can't imagine you'd see results very different from mine in your own machines' journals anyway.

```
      1 Oct 13 21:27:28 frys.old.box org.kde.krunner[18900]: RS_DEBUG::setLevel(3)
      1 Oct 13 21:27:28 frys.old.box org.kde.krunner[18900]: RS_DEBUG: Critical
      1 Oct 13 21:27:28 frys.old.box org.kde.krunner[18900]: RS_DEBUG: Errors
      1 Oct 13 21:27:28 frys.old.box org.kde.krunner[18900]: RS_DEBUG: Warnings
      1 Oct 13 21:27:29 frys.old.box librecad[18900]: QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
      1 Oct 13 21:27:31 frys.old.box org.kde.krunner[18900]: QG_LayerWidget::update: nullptr lastLayer
      1 Oct 13 21:27:56 frys.old.box org.kde.krunner[18900]: QG_LayerWidget::update: nullptr layerList
   2381 Oct 13 21:45:35 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   3847 Oct 13 21:45:36 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   4968 Oct 13 21:51:11 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   1270 Oct 13 21:51:12 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   6239 Oct 13 21:51:19 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
      1 Oct 13 22:13:41 frys.old.box org.kde.krunner[18900]: Unexpected token "mv" found at position 0.
    144 Oct 13 22:25:20 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   4356 Oct 13 22:25:21 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   1776 Oct 13 22:25:22 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   4795 Oct 13 22:26:07 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   1494 Oct 13 22:26:08 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   3935 Oct 13 22:27:57 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   2367 Oct 13 22:27:58 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   3601 Oct 13 22:30:19 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   2711 Oct 13 22:30:20 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   6333 Oct 13 22:42:21 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   5317 Oct 13 22:44:23 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   1026 Oct 13 22:44:24 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   4510 Oct 13 22:46:35 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   1843 Oct 13 22:46:36 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   1186 Oct 13 22:51:33 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   5195 Oct 13 22:51:34 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
    469 Oct 13 23:10:55 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   5981 Oct 13 23:10:56 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
    680 Oct 13 23:15:26 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   5780 Oct 13 23:15:27 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   2952 Oct 13 23:20:51 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   3518 Oct 13 23:20:52 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
    481 Oct 14 00:01:45 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   6039 Oct 14 00:01:46 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
      1 Oct 14 00:01:49 frys.old.box org.kde.krunner[18900]: Unexpected token "prop" found at position 0.
   6120 Oct 14 00:07:40 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
    398 Oct 14 00:07:41 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
    747 Oct 14 00:07:52 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   5772 Oct 14 00:07:53 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   4435 Oct 14 00:08:35 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   2083 Oct 14 00:08:36 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
   3142 Oct 14 00:09:03 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
[... 18 lines cut ...]
   2883 Oct 14 01:06:34 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
      1 Oct 14 01:15:27 frys.old.box org.kde.krunner[18900]: RS_EntityContainer::optimizeContours: hatch failed due to a gap
   4680 Oct 14 01:16:03 frys.old.box org.kde.krunner[18900]: RS_Modification::remove: no valid container is selected
[... cut ...]
      1 Oct 14 01:21:20 frys.old.box org.kde.krunner[18900]: QG_LayerWidget::update: nullptr layerList
```
I hope this report proves useful. Apologies if you find the narrative style of my bug reporting annoying, but that's just the way I roll. Thanks as always to everyone and anyone taking a look into the matter, and have yourself a good day.

### Operating System and LibreCAD version info

My Linux distro is **Mageia 8** (x86_64) and I'm using **Plasma 5** (KDE Frameworks 5.76.0
with Qt 5.15.2) as my desktop.

This bug report was produced with my distro's **LibreCAD-2.2.0-rc4**

DeliveryFry, 2022-10-18:
Wow. Okay, where to begin without getting too long winded…? I both agree and disagree with you.

Before going to bed last night, I checked in on this report again, and because it had been bothering me, did some quick eyeball math on the counts of the repetitions. They're so regular yet increase so gradually over time that, yeah, it occurred to me the total warnings might exactly equal the number of total entities on my active layer (or all the layers that were unlocked?) and that LC was iterating through *every single one* of them with each remove action. What you wrote above confirms that suspicion. Thanks. I'm guessing the entities are stored in an array or structure or something, and they constantly get iterated through over and over again? Am I right?

Anyway, by the time I had this insight, it was too late to test my theory and add a comment or edit my report, so I thought I'd save it for today, and yep, after conducting a simple experiment in a new drawing, I was able to verify that the number of warnings always seem to equal one less than the total of number of entities (on the active layer?).

But I disagree with you that this issue is somehow connected with the auto backup or that it has anything at all to do with `#outbuilding-e.dxf`. Those two `kf.kio…` messages are completely bogus and I could've (perhaps *should've*) expunged them from my journal extract and removed that file from my directory.

LC is *constantly* leaving those `#foo.dxf` files behind after it shuts down. They're just ordinary `#lock` files aren't they? Failing to clean up after itself may be yet another bug. Anyway, once in a while, when I notice that a bunch of them have piled up, and when LC isn't running, I delete them manually. In the case of the parent file, `outbuilding-e.dxf`, I haven't modified that one since 11 July. I did, however, load it up a few weeks back to compare something in that drawing to another, and that's corroborated by the timestamp on the lock (?) file: `#outbuilding-e.dxf` is dated 30 September.

As for the auto backup, I used to have it set for every 5 minutes, but that felt as if it was interrupting my work constantly, so I reduced it ages ago to every 15 instead. Apart from its expected freezes, I've never had a problem with it, and I'm 99% confident it has nothing to do with this issue. It's definitely *not* what's causing the excessive HDD activity I claim to be seeing, if that's what you're thinking. The auto backup causes the LED to *blink-blink-blink-blink-done!* Finishing very rapidly, even with my current huge project.

What I'm calling excessive is a constantly lit LED for the whole 2\~3 second duration of the remove action. I have next to no knowledge about how the Linux journal's infrastructure works or even how my distro has it set up for me, but what I suspect is that when a whole tonne of material is fed into it in quick succession a sync to HDD is triggered, perhaps as a defense mechanism in case of impending kernel panic or other catastrophe. After all, logs are only useful if there actually is a log to retrieve afterward. Apologies for all this crazy speculation, but that's all I got where the journal is concerned.

I can confirm that using your recommended `-d2` switch silences the warnings, and heck, starting it from the terminal as opposed to KDE's Krunner seems to keep *all* of LC's messages out of the journal entirely. I think what I'll do is add the switch to LC's `.desktop` entry and get back to you on whether or not it reduces, or altogether eliminates, the delay. Comparing it to the `move/copy` action, which never generate any warnings and complete nearly instantaneously — even when involving hundreds of entities at a time (as I mentioned in my original report) — I suspect it *will* help immensely. So I offer you a huge (and hopefully not premature) thank you for that. But I have to admit to being a bit confused: is this issue actually a *bug* then or not?

DeliveryFry, 2022-10-18:
All righty, I copied the main `librecad.desktop` to my own `~/.local/share/applications/` directory, added the `-d2` switch to the `Exec=` line, ran `update-desktop-database`, and fired up LC via Krunner as I normally would.

In a new drawing I created a long, arbitrary line at the origin, gave it 99 friends with the `parallel/offset` tool, used `rotate` at 15° on the lot of them to create a "flower" with 24 petals, then used `move/copy` on these first 2,400 entities three times across four unlocked layers.

So now with 9,600 potentially selectable entities at the ready, I zoomed way out, clicked at random until I'd grabbed 10 of them, pressed `[DEL]`, and… so far as my slow human time-sense is concerned, the removal action might as well have been ***instantaneous!*** :partying_face:

As expected, all remained quiet inside the journal, so naturally there was no unseemly disk activity. As a quick proof-of-concept, this looks *very* promising as an effective workaround to the issue.

I took an entity count with `selectall` on my big project as well. It's currently at 160,794 total, but since my journal entries showed between 6,000\~7,000 repeats of the warning for the 2\~3 second delays, I figured 9,600 would be plenty enough for a test.

Anyway, it might be a week or two before I've done enough real-world work to be sure and get back to you, but so far this simple switch option has my hopes set pretty high. Thanks again.

lordofbikes, 2022-10-18:
Knowing your auto save setting, I can confirm that this is not related to auto save. It was just a guess.
Leaving the `#foo.dxf` files untouched is intended by design. Those are the auto save files, no lock files. In case of a LibreCAD crash, they may contain a more recent version and save your day.
They are also kept when you close an unsaved file. In case of a mistake, you also have the chance of some saved efforts.
But closing a saved file, the `#foo.dxf` should be removed.
There are also `foo.dxf~` files, those are the auto backup files, which is a copy of the original file when it was opened.

A _POC_ only works when you use the original 2MiB file. I can't say, why there are only ~6000 warnings when the drawing has ~160k entities. This is possibly because of blocks/inserts, I guess an  insert is creating one message, but handles much more entities.

The messages a printed to `stderr`, so when you start LibreCAD from a terminal, this terminal swallow up the messages. When you start it from a GUI, the window manager of your distro decide where the messages are collected.

Because LibreCAD just prints the messages but don't save them to disc, we have a case of multitasking. The LibreCAD task prints the messages and the WM task collects the messages and writes them to disc when its buffer is full. This should not make the LibreCAD GUI irresponsible.

Just another guess is, that it is memory related. I don't know how much RAM a 2MiB DXF would consume and if there are multiple copies in memory when deleting. But on an older box, it may be short on memory and the disc activity is memory swapping. When the OS is swapping memory between RAM and HDD, this would block the LibreCAD task until the needed memory blocks are back in RAM.
This could explain the behavior, but as I said, just another guess.
You can use `top` in a terminal to see memory consumption and examine this.

Probably you can test the file on a more performant system to see if it matters. Test it with warnings enabled and disabled.

DeliveryFry, 2022-10-20:
Man, oh man, you've really given me lots to think about, experiment with, and comment on again.

**First:** regarding those `#foo.dxf` auto saves… Silly me! I've never actually bothered to look inside one to see what they actually contained. But your explanation certainly solves a bit of a mystery for me. It also brings up one of the few *minor* annoyances I have with LC that I haven't mentioned so far (in comparison with my earlier list of *most* annoying annoyances).

It's always bothered me that if I open a drawing, **zoom in or out** to check something — without making even the *slightest* change — LC will prompt for me to re-save it *again* because zooming alone is enough for it to consider the drawing modified. LC is the *only* software I can think of that I've ever seen do this. I don't know if you have some clever justification for why this behavior is necessary, but I think it's ridiculous. I always hit "close" without saving when I get those prompts. *But now you're telling me that **this** is the reason why I keep having all these auto-save files pile up in my directories?* ***\*\*\*Facepalm!\*\*\**** :facepalm:

Well, at least now I know how to get around that too… Does it merit another bug report of its own?

**Second:** I never suggested I thought the excessive HDD activity make LC's GUI unresponsive. Apologies if I somehow gave you that impression. Noticing it was just what prompted me to investigate what was going on. You'd already said in your first reply that it's the single-threaded iteration through all those entities blocking the GUI, and based on my own observations, I readily agree with you. Obviously the GUI only becomes accessible again once the remove action (that is, the iteration) finishes, and not surprisingly, the remove action happens to finish at *exactly the same instant* the journal is done capturing its warnings — regardless of whether the journal writes out to disk or not.

I agree this will take some real-world tests to verify, but I've still got high hopes that with the warnings silenced a remove action will now complete just as quickly as any other copy, move, rotate, or whatever action.

**Third:** Why there are only 6\~7k repeats of the warning when the whole drawing has 160+k entities has been bugging me since my first reply after the initial report, which is why I kept dropping in all those question marks. Does it iterate *all* entities? Only the entities on the *active* layer? Only *selectable* entities on all *unlocked* layers? Or maybe unlocked layers that are also *visible*?

To partly answer one of those questions, I can already tell you it has to be more entities than just those on the active layer. I'm pretty sure I know exactly what I was doing during that session my example journal was generated — I'd seriously borked a part of the drawing and was making some major corrections. But I checked, and the main layer I was working on only has something like 671 total entities. So there's something more to it. I had three layers unlocked and maybe another dozen visible. I'm pretty sure you're right, and it has to do with entities within blocks. One of the three layers contains several inserts of a block that itself has been replicated many times over with the rows/columns array feature. 

Anyway, if you don't have any more of an idea where those numbers came from than I do, a bit of experimentation might definitely be worth it. Honestly, I'm glad it wasn't repeating the warning 160+k times with every delete because I'm sure the delay would have been crippling. And that's one of the experiments I have planned. I'll make a copy of my project, unlock everything, grab a few entities to delete at random, and cross my fingers. I'll get back to you on how that works out.

**Fourth:** I don't have a "higher-performance" machine at my disposal at the moment, but I'm actually still very satisfied with this "good-enough-for-me" old box. And if you really *do* want to know more about it, sure, no problem. It's a 2008 HP-Pavilion with Intel Core 2 Duo CPUs @2.0GHz and 4GiB of DDR2 RAM, plus another 4GiB of encrypted swap space which barely gets touched because my `swappiness = 10`. Its graphics card is an old Nvidia GeForce 9600 with 512MiB VRAM. It's got two HDDs, both SATA mechanical. Oh, and my `/home` is fully LUX encrypted which presumably adds a bit of performance lag and memory overhead.

Now, here are some numbers I took note of for you: after a fresh boot and logging into a new KDE session, the system at idle has a total memory usage of \~575MiB. After starting KDE's Konsole terminal and loading up `top`, it reports, apart from itself, 187 sleeping processes. 

Then, starting LibreCAD via KRunner (using the `.desktop` file with the `-d2` switch) `top` reports its RES (which I believe is what you want to know) as 118,356 kiB. After closing my empty default template, it drops to 111,312 kiB. Finally, opening my 2 MiB project file… it increases to a mere 184,680 kiB. Hardly devastating so far as I'm concerned.

Out of the handful of programs I use daily to get "real work" done, the two biggest memory hogs are *LibreOffice* and *Firefox*, with Firefox being hands-down the worst of all offenders. Right now, as I'm typing this up, I have it open with four tabs, plus Konsole, Kwrite, Okular, and Kinfocenter. Those last two are because I was double checking my machine's stats for what I wrote above. According to `top`, Firefox currently has an RES of 475,980 kiB an [... cut ...] ,860 kiB. The PDF Okular has loaded is the 181 page, 5.7 MiB service manual for my computer. With all that, total RAM usage is just under 1.8 GiB. Swap is at 0. So yeah, in the grand scheme of things, LC is wonderfully light on its resource usage.

There you've had a peak into my old box while under normal load. Nothing exciting, and it's still plenty fast enough and capable enough for my purposes, and until it craps out completely or becomes too slow to be usable, I have no plans to replace it.

lordofbikes, 2022-10-20:
Setting a drawing _modified_ on zooming is because the viewport is part of the DXF file. So when the viewport change with zooming or panning, the drawing is modified from that point of view.
Iirc this was once disabled, but then it was impossible to intentionally save a drawing with a specific view. That annoyed other users who need this and thus it was restored again.

A quick test confirmed my assumption. Place 5 inserts in an empty drawing shows 4 warning when deleting one of them. Select all and explode them results in more than 100 entities and thus more than 100 warnings when deleting a single entity from previous 4 blocks.
In your case, selecting all and explode, delete will end up with ~160k warnings.

The RAM/swapping guess was just to catch at a straw :wink:
I feared that your box has enough memory for a 2MiB DXF, even when it is a bit older.
Especially because the binary representation in memory was expected to be a fraction of the ASCII DXF file size.

DeliveryFry, 2022-10-28:
Hi again. This past Saturday I conducted a few experiments, but I've been too busy this week to properly comment on them. I'll gladly provide more info if you're interested, but suffice it to say that silencing the warnings does indeed completely eliminate the delay that prompted me to file this report.

In brief: with warnings enabled, after unlocking all layers in my big project and exploding all blocks, deleting any 5 entities at random results in a delay of \~11 seconds. With warnings silenced, the same remove action finishes nearly instantaneously.

Based on this, I would definitely recommend considering what you'd mentioned at the very start…

> From a quick look I'm not sure if this message is worth a warning, as it seems to be triggered for each not selected entity.

…and removing, restructuring, or whatever, that warning. I'll continue running LC with that `-d2` switch you suggested until this gets properly resolved, but would prefer to keep warnings enabled by default as that seems most sensible to me.

lordofbikes, 2022-10-29:
Many thanks again!
I'll check for the best option, using a level above the default or removing it at all.

commit dbc1f410 (2022-10-30): fixed #1567, flood of RS_Modification warnings
closed 2022-10-30 by lordofbikes