LibreCAD/LibreCAD #1569: Why this file (.dxf) blocks the LibreCAD program?

GrzegorzFK, 2022-11-03:
Very strange. This is a simple drawing.dxf, too difficult to work on it. 
This file opens and after a while the program stops working.

Gregory

[

### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

[p2-INOX-Wanna-Bok-L.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/9933162/p2-INOX-Wanna-Bok-L.dxf.zip)

bajraan, 2022-11-04:
Hi Grzegorz,

I can easiliy working with your file. 
I used and edited your file for 120s without any problems.
Below you can find edited file by me. 

LibreCAD:
Version: 2.2.0-rc4-21-gdbc1f410
Compiler: GNU GCC 12.2.0
Compiled on: Oct 30 2022
Qt Version: 5.12.11
Boost Version: 1.75.0

PC:
System Type			x64-based PC
OS Name				Microsoft Windows 10 Pro
Version				10.0.19044 Build 19044
Installed Physical Memory (RAM)	8,00 GB
Proccessor 			i5-3450 CPU @ 3.10GHz   3.30 GHz
BaseBoard Product		H61M-S1

Grafics Card
Name				NVIDIA GeForce GTX 960
Driver Version			31.0.15.1748
Resolution			1920 x 1080 x 60 hertz
[p2-INOX-Wanna-Bok-L_20221104.zip](https://github.com/LibreCAD/LibreCAD/files/9936927/p2-INOX-Wanna-Bok-L_20221104.zip)

GrzegorzFK, 2022-11-04:
Hi, thank you for the support.
I was opened your file, next,  rotated it 90 degrees, and ... the program
stopped working.

But, thanks to you, I know it's my computer's problem.

Best regards,
Grzegorz
---------
LibreCAD
Wersja: 2.2.0-rc4
Compiler: GNU GCC 12.1.0
Skompilowano: Jun  7 2022
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

GrzegorzFK, 2022-11-04:
[image]

bajraan, 2022-11-04:
Rotating is also working,

Below are my ideas about what you can check to find what causing your LibreCAD crashing
1) You can check different .dxf files if LibreCAD crashing.
2) You can check the Operating System using different Operating System for example Linux Ubuntu.

Ubuntu can be started from bootable USB stick (you do not have to install it). You have to only prepare bootable stick with ubuntu image using for example bolena etcher.

IMHO discussion should be continued on the forum.

Best regards,
Bajran

GrzegorzFK, 2022-11-04:
Hi. This file, which crushing the LibreCAD.exe works properly in different CAD programs.
What`s more, this file is only one a dxf file from many, which do it on my computer!
I don`t know : WHY

dxli, 2023-05-08:
I'm testing this dxf file running the latest master branch v2. 

Cannot reproduce this issue.

closed 2023-06-06 by dxli