LibreCAD/LibreCAD #1571: Error in compilation with last build

fa201, 2022-11-05:
Today, I saw a code update to I pulled from the repo and built from source as usual. I got an error when doing `make -j4` on Xubuntu.

Here are the last lines of make command:

[image]

Translations:
"échec de l'édition de liens finale" -> "failed final link edition"
"on quitte le répertoire" -> "leaving the directory"

Thanks.

lordofbikes, 2022-11-06:
This code built fine with *GitHuc Actions*, so I assume an issue with your environment.

When was the last successful build?
You can `checkout` the previous version, which built correct, to see if this still builds.
If not, there was eventually an OS update causing this.

Which Qt and GCC versions do you use?

Please paste the complete `ld` command which triggered the error. The screenshot only shows the last few parameters, there mus be more. The command starts with `ld` or `/usr/bin/ld`.

fa201, 2022-11-06:
I do not remember when the last build was done: around 2 weeks I guess.
Building version from Oct 30th is still NOK, and yes I did get some OS library update last week...
qmake 3.1 using qt 5.15.3
gcc 11.3.0
I add the complete output for the build attempt.
[output.txt](https://github.com/LibreCAD/LibreCAD/files/9945950/output.txt)

lordofbikes, 2022-11-07:
So this seems somehow triggered by [Boost](https://www.boost.org/).
Which version of Xubuntu do you have?
Can you also find out the Boost version?

fa201, 2022-11-07:
Xubuntu 22.04 LTS.
I guess libboost-dev -> 1.74.0.3

fa201, 2022-11-08:
I did not know this command (I made a search in aptitude package list with the interface). 

It shows 2 packages : 
`libboost-dev/jammy,now 1.74.0.3ubuntu7 amd64  [installé]`
`libboost-dev/jammy 1.74.0.3ubuntu7 i386`
This 32-bit package is coming out of nowhere (I never install this type of package) and it is not shown as installed in aptitude. 

I gave a try with `make clean` and then a regular `qmake` and `make`and it works !
Thank you very much @lordofbikes 

I removed tags and closed the issue.

closed 2022-11-08 by fa201