LibreCAD/LibreCAD #1572: Area and hatch incorrect results using semi-circles

jophut, 2022-11-08:
### Expected behavior
Area of the quarter circles should be a quarter of a total circle area.

### Observed behavior
Area of a circle in the fourth quadrant "D" is correct.
Area of the quarter circles the first, second and third quadrant's E, F and G are wrong.

### Steps to reproduce or sample file
[20221108_LibreCAD_Area_wrong.zip](https://github.com/LibreCAD/LibreCAD/files/9962295/20221108_LibreCAD_Area_wrong.zip)

[image]

### Operating System and LibreCAD version info
Windows 10 Home, Version: 21H2
LibreCAD 2.2.0-rc3

bajraan, 2022-11-09:
Easy to reproduce this behavior on my setup:

LibreCAD:
Version: 2.2.0-rc4-21-gdbc1f410
Compiler: GNU GCC 12.2.0
Compiled on: Oct 30 2022
Qt Version: 5.12.11
Boost Version: 1.75.0

PC:
System Type x64-based PC
OS Name Microsoft Windows 10 Pro
Version 10.0.19044 Build 19044
Installed Physical Memory (RAM) 8,00 GB
Proccessor i5-3450 CPU @ 3.10GHz 3.30 GHz
BaseBoard Product H61M-S1

[image]

dxli, 2023-05-27:
1. Area of non-manifold is tricky for our algorithm which is based on Green's theorem;
2. The 90 degree section trouble is a separate bug which should been fixed in the master branch. The bug was due to signs of reversed arcs were not considered in the initial implementation.

dxli, 2023-05-28:
60817734 fixes cases E/F/G in [20221108_LibreCAD_Area_wrong.zip](https://github.com/LibreCAD/LibreCAD/files/9962295/20221108_LibreCAD_Area_wrong.zip)

Case C is harder: The main reason, LibreCAD doesn't have a topology layer yet. It means LibreCAD couldn't reliably determine which area is the inside of the contour, if the contour is not a simple closed contour.

The workaround, always find the area using simply closed contours:
1. Contour bonndaries are all cut at intersection points;
2. There's no lakes/islands/islands in lakes.

An actual solution: add topology support for hatching.

fa201, 2023-05-29:
@feragon Could this happen on V3 also ?

dxli, 2023-06-15:
Give closed contour loops, topological relation can be figured out: if there's no intersection (tangential points not considered yet) between loops. 

I will give it a try.

V3 chose to workaround by an appropriate algorithm, but in v2, we tend to rely on closed form solutions.

dxli, 2023-07-15:
Add a test case:
[hatch1.zip](https://github.com/LibreCAD/LibreCAD/files/12061078/hatch1.zip)

dxli, 2023-08-12:
Add another test case:
[hatch3.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/12326501/hatch3.dxf.zip)

closed 2024-05-18 by dxli

jophut, 2024-11-23:
Using LibreCad v2.2.0

And this dxf-file:
[20241123-2045.dxf.zip](https://github.com/user-attachments/files/17882415/20241123-2045.dxf.zip)

Unfortunately hatching the same contour 8 (copies) times, gives 8 different areas:
2299280
2253427
2252616
2246034
2222998.9
2237881
2230771
3548551

While, to my calculations, the only correct area should be 2093719.7

dxli, 2024-11-23:
Please try 2.2.1  release candidate, or 2.2.2

It's a known issue for 2.2.0