LibreCAD/LibreCAD #1574: This Program is Slow

nickrazes, 2022-11-14:
This program is too slow at the moment. This issue is a request for others to look into it and/or fix it. See attached *.dxf file (note that it is zipped and you need to unzip it and then open it with LibreCAD). Notice that trying to do anything with this file is very slow. Even zooming in and out. I know there's a lot of stuff in it, but AutoCAD can handle a file with 10x as much without a problem. Any ideas?

Expected behavior: Expected to be fast enough to be usable with large files.

Observed behavior: Observed to be slow with files that, in reality, aren't all that big. It works fast with very small files but once there begins to be a lot of elements in it, the program starts getting more and more sluggish to the point of being unusable.

Steps to reproduce or sample file: See attached sample file.

Operating System and LibreCAD version info: Operating system is Windows 10. LibreCAD version is 2.1.3. 

[librecad.zip](https://github.com/LibreCAD/LibreCAD/files/10000564/librecad.zip)

lordofbikes, 2022-11-14:
This file has nearly 1 million entities, LibreCAD was never intended to handle such files.
This would need a complete revamp of internal organization and handling of entities, what is far above our capacities.
Maybe LibreCAD 3 can handle this in future.

closed 2022-11-14 by lordofbikes

nickrazes, 2022-11-14:
1 million isn't that much for a computer.

Personally, I think that, even if it's unplanned, that you should've just left it open and just let it sit. 

Nobody wants to use LibreCAD to work on some large personal project of theirs only to find out later on that the program can't handle it......

nickrazes, 2022-11-15:
One more thing. I think before you go about completely revamping the internal organization and handling of entities, that you should at least consider adding in smart-zoom-breakpoints. You'd hate to do all that work only to find out that it doesn't work in the end. I think AutoCAD is getting files 10x the size of the one I attached to work by simply not loading/reloading everything on every zoom. For example, when you first open the file I attached, since it's so zoomed out, you don't need to load every little bit and piece of information since a monitor doesn't have enough pixels to display it all anyways. Instead, you'd have a smart-zoom-breakpoint that realizes this and loads a smaller/faster dummy version first. Then, as you zoom in, it'll eventually load a little more information, but only what you can see on the screen and some to each side but keep the dummy version for views further away from where I'm viewing (in case I zoom out or pan). Hope it makes sense. I'm sure you get the idea I'm suggesting. You'll also need a regen command so if a user needs to see more information then what it's showing you can make it do so. I'm pretty sure AutoCAD handles this file that way since, when it does reload, it takes forever like LibreCAD does and when you zoom in really far, you can see that it's a dummy version that's not displaying everything perfectly until you regen it. Anyways hope it helps. Good luck with your program.

fa201, 2022-11-15:
@nickrazes 
With last build and a computer with only an average power without GPU, I can zoom in/out in less than 2 seconds. But on linux OS.
Then I copied 10 instances of the complete drawing: about 5 seconds. Then zoom out: about 7 seconds.
Not that bad for a non-professional-grade application!

nickrazes, 2022-11-16:
I agree it's not THAT bad for a non-professional-grade application.

My computer is an average power without GPU as well (give or take). On my computer it takes about 20sec to open and 14sec for each zoom. It's interesting it's different for you.

It is a legitimate issue though which I don't agree should be closed like this. It's a red flag which sends the message: "hey, we're cool with slowness here."

Basically, since there's no functionality to help handle large files, it means that if you're using it for a larger project with many components that, over time, your project will get slower and slower and slower as you add more and more stuff. This needs some work IMO, that's all I'm saying. I'm not trying to bash it or anything. Maybe I should've worded it different.

lordofbikes, 2022-11-16:
> My computer is an average power without GPU as well (give or take). On my computer it takes about 20sec to open and 14sec for each zoom. It's interesting it's different for you.

With the provided information it is hard to say what makes the difference.
Probably start with using a newer version. We are near to a new release and it is safe to use latest continuous build.
Iirc, there were some improvements concerning performance since 2.1.3.

> It is a legitimate issue though which I don't agree should be closed like this. It's a red flag which sends the message: "hey, we're cool with slowness here."

Closing like this is just because this debate is closed long ago. It has nothing to do with coolness. A simple search on Google with keywords **librecad** **slow** would have led you direct to our forum where you find the discussions and support from experienced users.

nickrazes, 2022-11-16:
Yes, the newer version helps. Each zoom is 2sec now, same as fa201. Certainly better. At least you are indeed working on it. 

With all do respect, sir, searching google for "librecad slow" doesn't turn up any solutions that actually fix it completely. It's just people talking about it and agreeing that it's slow. It's hardly a "debate".

There's also no "librecad slow" threads on here, and there should be one in my opinion. It's kind of hard for a community to work on improving it together without one......

Coolness? What? Closing it like this communicates that you don't want me working on it; you don't want anyone in the community working on it; you're not working on it (even though you apparently are since there's improvements from one version to the next); etc.

At any rate, just go ahead and leave it closed. Whatever. I lost interest in trying to help anyways and no one wants to spend a bunch of time trying to improve things just so their pull request can get blocked in the end.

This conversation is really rubbing me the wrong way so I'm going to take my leave. I was just trying to help. There was no need to get defensive about it. I wish you nothing but the best of luck with your program. Maybe I'll check back LibreCAD 3. Maybe. Goodbye.