LibreCAD/LibreCAD #1575: "save" turns into save as on a CIFS share

tu77, 2022-11-22:
### Observed behavior:
If I open a existing dxf file on my Samba share and try to save my work, it will save this file only once, on the next save it comes up with a save as dialog and tries to create a new file.
LibreCad Comand line shows a access error for the file now.
wit a local file this behavior will not occur.
 
### Expected behavior:
It should just save the file

### Steps to reproduce or sample file
-Open a file 
-make some small changes
-save.. no porobem
-make some more changes 
-save... save as window opens..

### Operating System and LibreCAD version info
Linux Mint 21
tested all versions starting by 2.0 till latest 2.2 RC4-23 with same problem
Tested it on different computers with the same result

I think I`s a problem with the cifs share but I did check all settings in Samba and CIFS mount 
LibreCad is the ony Program I observed such problems.
Problem occurred not in Linux Mint 20

lordofbikes, 2022-11-22:
The `Save as...` window is just to give you the chance to save your work and avoid loss of effort.
It is triggered after generic `Save` has failed.

You already located the problem:
> LibreCad Command line shows an access error for the file now

That means the original file is locked somehow and it is impossible to write to the current file.
I see nothing else that LibreCAD can do when save fails. Only chance to save changes then is `Save as...`.

So the challenge is to find out why the original file is locked.
It is not clear where this CIFS share is located. Is it hosted on the mentioned Linux Mint system?
Is it on a NAS or a Windows share?

This sometime happens, when another application locks the file. I know this behavior when the file is included and opened in OpenSCAD at the same time. Then OpenSCAD sometimes locks or modifies the DXF and changes in LibreCAD must be `Saved as...` too.

You went back to LibreCAD 2.0.x without any difference, thus it is obvious that there were no changes in LibreCAD which may have triggered this issue. I can't imagine, that this would have been undetected such long time.
The obvious change you mentioned is the Linux Mint version, which changed from 20 to 21.
Are there any other recent changes in your environment?

When this worked with Linux Mint 20 and arose with Mint 21, you probably should report this to the Mint maintainers too.

tu77, 2022-11-24:
Hello, 

I did some more investigation and it turned out , that the CIFS/SMB protocol version 3 is the problem.
My server is running on Ubuntu and only on clients based on Linux mint 21 I got this problem.
Linux mint 20 has an older CIFS version than 21, not sure what exactly is causing this problem, maybe its the implementation of ACL's in the Protocol version 3 .
I'm not really happy downgrading the Protocol version but it solved the problem for now.
It is still a bit strange to me, that LibreCad is the only Program showing this Problem.

so finally changing "vers=3.0 to vers=2.0" in the fstab file on the client system solve the problem!
Note; the Samba server should allow at least Version 2, if it is restricted to version 3, this will not work.