LibreCAD/LibreCAD #1576: PDF export doesn't seem to include dimensions or blocks

dgluss, 2022-11-29:
### Expected behavior
PDF export should match printed behavior.

### Observed behavior
Dimensions and blocks are not there.

### Steps to reproduce or sample file
Load enclosed dxf file. Print, and also export to PDF.

### Operating System and LibreCAD version info
LibreCAD Version: 2.2.0-rc4-23-ge3fae3dc
Compiler: GNU GCC 10.2.1
Compiled on: Nov 28 2022
Qt Version: 5.15.2
Boost Version: 1.74.0
System: Debian GNU/Linux 11 (bullseye)
Current unmodified Git clone. 
[mountain lane.dxf.gz](https://github.com/LibreCAD/LibreCAD/files/10108883/mountain.lane.dxf.gz)

lordofbikes, 2022-11-29:
This is presumably because of default line width. See #1326

It also depends on the PDF render library. For me PDF export produced valid results, it contains dimensions and blocks.

But, layer line width setting is default (0.0) and block lines are set to `By Layer`.
Dimensions have line width settings in `Current Drawing Preferences` and are usually set to `By Block` which becomes `By Layer` too.
Other lines have a fixed width set to 0.40mm.

So change layer `0` line width to another value should export dimensions and blocks.

The basic issue with default line width is already tracked in #1326, so I close this one.
If line width is not the solution for your issue, feel free to report more information and reopen it.

closed 2022-11-29 by lordofbikes

dgluss, 2022-11-30:
The definitive test for this is to take the pdf file and run pdf2ps on it. Indeed, the Postscript file renders correctly (with the dimensions and all) so there is no problem with LibreCAD. Thanks for looking at this bug.

sebma, 2024-09-11:
I still have this problem on LibreCAD v2.2.0.2 on Ubuntu 22.04.

Is there a way to choose another PDF render ?