LibreCAD/LibreCAD #1577: Zoom-Pan tool button stays active after <ESC>

lordofbikes, 2022-12-01:
### Expected behavior
Uncheck the active `Zoom Panning` tool button or push `ESC` should disable the pan function and show text again.

### Observed behavior
Deactivate `Zoom Panning` with tool button does not work at all.
Using `ESC` deactivates panning, but text stays there as outline boxes.

### Steps to reproduce or sample file
- open a new file
- place a text anywhere
- click on [image] or use menu `View` -> `Zoom Panning`
- move the drawing around with the mouse
- again click [image] or use menu `View` -> `Zoom Panning`
- panning is not deactivated
- press `ESC`
- panning is cancelled, but text stays as outline box

### Operating System and LibreCAD version info
OS independent, latest code 2.2.0-rc4-23-ge3fae3dc

Workaround from forum members is to close and reopen the drawing.

Another option is to use the mouse. The middle button is also `Zoom Panning`, but working correct.
So use the mouse generally for now or at least to disable `Zoom Panning` tool button or menu function to make text visible again. 2-button mice usually have a middle button simulation by pressing left and right button together. 

cross-link: https://forum.librecad.org/Only-a-box-appears-when-entering-text-td5722083.html#a5722228

gregochan, 2023-04-11:
same issue with me, text display just fine, and once after the zoom panning, all turned into boxes and no way to get it back other than closing the file and reopen it.

dxli, 2023-06-06:
The issue was caused by missing of a customized finish() method for RS_ActionZoomPan.

Fixed in master

closed 2023-06-06 by dxli

RaPaDa5, 2023-06-12:
> ### Expected behavior
> Uncheck the active `Zoom Panning` tool button or push `ESC` should disable the pan function and show text again.
>
> ### Observed behavior
> Deactivate `Zoom Panning` with tool [...]

There is actually a way how to discontinue zoom panning and make the text (including dimension) to reappear: Whether you run zoom panning by clicking on the respective icon or by a CMD line shortcut "zp", you can exit from the process by the right click. (ESC does not work as the ticket above says.)

1holdstrong, 2023-10-29:
Hi, Have followed the #1577 Github issue and got the impression the issue was solved.  Have just upgraded to Version: 2.2.0.2, Compiler: GNU GCC 13.1.0, Compiled on: Jul 29 2023, 
Qt Version: 5.12.11, Boost Version: 1.75.0 and problem (text showing as rectangles after using pan) still occurring.  Work around to not use pan or close & reopen drawing still ok.
Apologies if I’ve misunderstood the forums.