LibreCAD/LibreCAD #1581: Publish a x64 version for Windows

jd-imi, 2022-12-19:
With the release of LibreCAD 2.2.0,  LibreCAD-Installer-2.2.0.exe  is an installer for the x86 version of Windows. It can be nice to have also a x64 version.

fa201, 2022-12-21:
You cannot use the x86 on 64-bit OS ?

bansan85, 2023-01-01:
I just follow instruction from [git ](https://github.com/LibreCAD/LibreCAD/tree/master/.github/workflows)

If you are a developer, you can very easily build an x64. I just tested it.

Releasing an x64 version of LibreCAD could be fine.

dxli, 2023-06-05:
I feel we should support CI for x64 first.

looks like we only need a few changes in our building scripts:

mingw73_32 to mingw73_64

For NSIS, InstallDir from "$PROGRAMFILES/LibreCAD" to "$PROGRAMFILES64/LibreCAD"

dxli, 2023-06-07:
5406cbda : added CI support for windows x64

using Qt LTS 5.15.2

bansan85, 2023-06-08:
@dxli Thanks :)

I just download the installer from asset https://github.com/LibreCAD/LibreCAD/actions/runs/5207737880

Task manager says that Platform of Librecad is 64 bits :) and I successfully open my draw.

But the default installation path is still "C:\Program Files (x86)\LibreCAD\". You should remove " (x86)".

jd-imi, 2023-06-08:
> @dxli Thanks :)
>
> I just download the installer from asset https://github.com/LibreCAD/LibreCAD/actions/runs/5207737880
>
> Task manager says that Platform of Librecad is 64 bits :) and I successfully [...]

I don't know if it's an expected behaviour or not but I think that if you uninstall LibreCAD first and reinstall the 64 bits version, it will install in "C:\Program Files\LibreCAD".

In `LibreCAD/scripts/postprocess-windows/nsis-5.4.nsi`, we can find:
```
  ;Get installation folder from registry if available
  InstallDirRegKey HKCU "Software\LibreCAD" ""
```
And on my laptop I do have `C:\Program Files (x86)\LibreCAD`

bansan85, 2023-06-08:
I already tried to uninstall/reinstall but `C:\Program Files (x86)\LibreCAD` is still the default.

But I noticed that `HKCU "Software\LibreCAD"` is not deleted when uninstalling.

I manually deleted `HKCU "Software\LibreCAD"` and now the installer has `C:\Program Files\LibreCAD` as default path.

I think that `HKCU "Software\LibreCAD"` should be deleted when uninstalling.

jd-imi, 2023-06-08:
IMHO, @dxli implemented an even better solution (using another registry key for the x64 version) ! :-)
https://github.com/LibreCAD/LibreCAD/commit/c9cb67404b8d77207886095268860aed9bc7c13f

bansan85, 2023-06-12:
I tried with assets from https://github.com/LibreCAD/LibreCAD/actions/runs/5240592278 based on commit b2ffe4410cf49fdc6fa76111d6adc985a6e210e5

It's a bit better.

I installed x86 version. Default path : "C:\Program Files (x86)\LibreCAD"

I installed x64 version. Default path : "C:\Program Files\LibreCAD". LibreCAD x86 still installed but replaced in "add/remove program" by x64 version. It's should not but not really important.

I uninstall both version.

I installed x64 version. "HKEY_CURRENT_USER\Software\LibreCAD" is created and not "HKEY_CURRENT_USER\Software\LibreCADx64". It's strange since your commit looks good.

dxli, 2023-06-15:
Windows x64 CI builds work for me now.

closed 2024-05-18 by dxli