LibreCAD/LibreCAD #1583: Issue with dimensions reported through the forum

fa201, 2022-12-28:
https://forum.librecad.org/quot-Cutting-Lines-quot-Bug-td5722327.html

lordofbikes, 2022-12-29:
This issue is triggered by dimension length 0.0.
It arose when abusing aligned dimension as leader.
An aligned dimension needs different start and end point for various angle calculations.
This is not given with a 0.0 length dimension. Calculating angles is impossible.

As the transformation happens on save/reopen, the issue can be located in [LibreCAD/libdxfrw](https://github.com/LibreCAD/libdxfrw) or it is caused by above mentioned impossible angle calculation.

Anyhow, as this issue does not appear on regular aligned dimensions > 0.0 and impossible calculations for 0.0 length dimensions, best solution would probably be to prohibit the creation of 0.0 length dimensions at all.

fa201, 2022-12-29:
Thanks for the explanation, I was not sure how to name the issue but I did not want to miss it in the list.