LibreCAD/LibreCAD #1585: Hatches at angle are specified/handled in deg, but stored in rad -> wrong angle after re-open

streumix, 2023-01-03:
Hi,

I've done some trivial sketches using hatches at an angle, e.g. iso03w100 at 30° 45° or 90°.
But when re-opening the drawing after save (I've tested all DXF variants >= R14, which intend to store hatch information) the angle of all hatches is changed == wrong.

### Expected behavior
Hatch angle should be retained in file storage.

### Observed behavior
The value specified for angle in deg seems to be converted to rad during save OR read, and this value treated as deg after re-open. A hatch with angle set to 90° and display as expected before close/save, turned into 1.57 (=pi/2) after save>close>re-open.  
So it's obviously a deg <> rad unit conversion issue in either save or read. 

### Steps to reproduce or sample file
Draw a trivial rectangle, hatch with pattern at a certain angle, e.g. iso03w100 @ 90°. Save, close, reopen, and measure the angle of the hatch. Angle has been converted to rad unit, but treated as deg for display. 

### Operating System and LibreCAD version info
Latest distribution supplied Librecad 2.2 on Fedora 36 x86 / 64-bit, kernel  6.0.15-200.fc36.x86_64

lordofbikes, 2023-01-03:
This seems to be a duplicate of #1545, which is solved and close.
I tested this and it still works for me with this code base.

This seems still to be an issue of Fedora and possibly its sub branches.
You should report this to your distributions maintainer.

The cause for #1545 was an erroneous patch on [libdxfrw](https://github.com/LibreCAD/libdxfrw), which is probably still around.
The latest version of [libdxfrw](https://github.com/LibreCAD/libdxfrw) is tagged with the label [LC2.2.0](https://github.com/LibreCAD/libdxfrw/releases/tag/LC2.2.0) and does not have this issue.

streumix, 2023-01-04:
Thanks for pointing me to #1545. Though I've searched for bugs like this, I didn't find this one. Maybe I had been to specific to hatches. 
Anyway, Fedora 36 comes with libdxfrw 1.0.1, whereas latest Fedora 37 is shipping version 1.1 by default. So I'd better pull in my already planned system upgrade. Maybe it solves my problem with libdxfrw.
Regards!

closed 2023-01-04 by streumix