LibreCAD/LibreCAD #1587: Skip first coordinates entry for relative operation

Jim-2249, 2023-01-10:
Feature request:
When stretching, moving, or maybe drawing a line, currently, one might enter the first coordinates as 0,0 followed, say, by 1,1. Often the first coordinates are unnecessary.
I request that when the first coordinates are entered, if they are prepended by an "@", that coordinate would be sufficient for the operation, and relative, eliminating the need to type two coordinates.
This could save repeated typing of "0,0"

### Desired behavior
- "mv @4,3" would move the selected object(s) 4,3 units.
- "line @6,3" would draw a line from the relative zero position to relative zero position + 6,3 units. This can currently be achieved by "line @0,0 @6,3"
- "stretch" [select box] "@-5,0" would stretch by -5,0 units

dxli, 2023-06-14:
Then, "line @4,3" is used in both cases:

1, a line to @4,3;
2, a line to start at @4,3

dxli, 2024-09-19:
", @4,3" from relative zero to "@4,3";

If skipped, it means relative zero.