LibreCAD/LibreCAD #1588: Saving file causes print scale issue

dnlombard, 2023-01-16:
### Expected behavior
File->Save does not corrupt print scale

### Observed behavior
Attached is viewed on screen. Page size is "Letter/ANSI A". Select File-Print Preview or select Print Preview from toolbar.  Observe scale is 1:1 in the print scale option menu in the toolbar. This is different from the maximum printable scale, 1.63268:1 for attached drawing. Also observe page rectangle is correctly sized relative to drawing. Exit Print Preview. Select FIle->Save or Save from toolbar. Select File-Print Preview or select Print Preview from toolbar. Observe print scale is still shown in the print scale option menu as 1:1, but  the page rectangle is now consistent with the maximum scale. Also observe that the maximum scale is no longer included in the print scale option menu. Select 1":1" from print scale option menu and observe that page rectangle changes to expected (larger for attached drawing) size.

While the above discusses Print Preview, the same behavior is observed when the file is printed, i.e., the drawing is printed at the maximum scale (1.63268:1 for attached drawing).

It may well be that Print Preview is intended to always change the scale to maximum and the error is the print scale option menu is not correctly updated.

### Steps to reproduce or sample file
[EMonk-Offsets.zip](https://github.com/LibreCAD/LibreCAD/files/10429759/EMonk-Offsets.zip)

### Operating System and LibreCAD version info
Version: 2.2.0-alpha
Compiler: GNU GCC 12.2.1
Compiled on: Sep 24 2022
Qt Version: 5.15.7
Boost Version: 1.76.0
System: Fedora Linux 36 (KDE Plasma)

fa201, 2023-01-26:
I cannot reproduce this. Fitting the drawing on page shows a scale of 46.05:1.
Version: 2.2.0-alpha
Compiler: GNU GCC 11.3.0
Compiled on: Jan  7 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Ubuntu 22.04.1 LTS

The _Print Preview_ is done before printing but it depends on the paper size selected in _Current Drawing Preferences_ which may be not saved to the DXF file. Then it looks normal to set the _Print Preview_ for each LibreCAD session.

gerardbm, 2023-04-16:
I cannot reproduce this. Actually, this was a known issue that was fixed in an older version.

dxli, 2024-05-18:
We have 2.2.2_alpha, 2.2.1 release candidates, and 2.2.0 stable releases. 

Closing this one on 2.2.0_alpha

closed 2024-05-18 by dxli