LibreCAD/LibreCAD #1589: Imported image absolute path

PeterHagen, 2023-01-26:
When an image is imported to a drawing, it's stored with its absolute path of the filesystem. This will make the image disappear when the drawing is opened on another machine, or another user account. The path also can't be edited in the property dialog.

### Expected behavior
Storing the image path relative to the drawing

### Observed behavior
Storing the image path absolute to the filesystem

### Steps to reproduce or sample file
Just insert some image, in my case a SVG, and check the properties

### Operating System and LibreCAD version info
macOS m1, Ventura and LibreCad Version: 2.2.0-2-gf55f341e

fa201, 2023-01-26:
A work around could be to delete the absolute path in DXF file, providing image and DXF file stay in the same folder. I checked it works. To make it easy to spot, you can search the DXF file for the image name with a text editor.

dellus, 2023-01-26:
If you place the image in the same folder as the dxf file, the image will always be found, where ever it was when you inserted it into the drawing.

PeterHagen, 2023-01-27:
The issue is that the image is inside a block, that is being added to different drawings in different directories. It would be helpful if the path of the file is editable in the property window. Now its a readonly element.

dxli, 2023-08-03:
The image source is now editable in the property dialog

closed 2023-08-03 by dxli