LibreCAD/LibreCAD #1591: Wrong translation license

comradekingu, 2023-02-12:
https://hosted.weblate.org/projects/librecad/librecad/#information says
GPLv2-only, whereas
https://github.com/LibreCAD/LibreCAD/search?q=any+later says
GPLv2-or-later.

It can be fixed in
https://hosted.weblate.org/settings/librecad/librecad/
https://hosted.weblate.org/settings/librecad/plugins/
https://hosted.weblate.org/settings/librecad/glossary/

If you add "kingu" as an admin I can look for other things to fix :)

lordofbikes, 2023-02-12:
Many thanks @comradekingu for pointing this out.
Is this related with my request for free hosting?

Our main license is https://github.com/LibreCAD/LibreCAD/blob/master/LICENSE, which is GPLv2 only.
This is based on the fact, that our code base is based on QCAD community code, which was GPLv2 when the fork started.
The files found by your search term are completely new files, which were added by contributors and are not based on QCAD code base. Earlier the files were named following the QCAD scheme, but on newer files we use the prefix lc_ to mark them as LibreCAD property. Most of the found files are lc_* files.

I'm not a lawyer, and most of these processes happened before I took over responsibility for the project. So when I created the Weblate LibreCAD components I used the most restrictive license, as I don't know, if there are any translations originating from QCAD base and if it is allowed to re-license them. Also, in translation files we can't separate QCAD base and LibreCAD additions.
I just know, that Ries, the founder of the LibreCAD fork, had discussions with the QCAD developer, and he was not willing to allow re-licensing the code for LibreCAD.

comradekingu, 2023-02-12:
Does that mean https://www.qcad.org/en/qcad-downloads-trial which says
https://github.com/qcad/qcad/search?q=any+later for things in "core"?
Seems pretty clear cut to me.

The license file in LibreCAD isn't specifically GPLv2-only
https://github.com/LibreCAD/LibreCAD/blob/master/licenses/gpl-2.0.txt#L242-L246

lordofbikes, 2023-02-12:
@comradekingu you link to current QCAD version 3.x and its code base.
But this is not the version from which LibreCAD was forked.
LibreCAD was forked in 2010 from QCAD 2.x community code, which is GPLv2 licensed, not _or any later_.

And the quote from the license just states, that the FSF can publish newer versions of GPL and that it may contain the phrase _or any later_, which is not the case for QCAD 2.x code and thus for LibreCAD too.

I still wonder why we discuss this, I believe that GPLv2 is an accepted license for free hosted Weblate.

comradekingu, 2023-02-12:
I think there is a misunderstanding here.
Having something GPLv2-only makes it incompatible with GPLv2-or-later, which would require every translator to relicense.
There is nothing in LibreCAD that actually says GPLv2-only, and GPL is a full and strong copyleft license, so it extends to the full work (with some possible disagreement over whether linking can onstitute "a work" a two non-separate entities in copyright.)

It doesn't matter what point the fork occurs (or what it is called), if it is a continuation of a work, it would have to be compatible (or later) to make it to GPLv3-or-later. I think that makes it compatible whether re-licensed specifically or not. Was it ever specified as "only" btw?

GPLv2 with "or later" specified in any of the files is just as much "GPLv2-or-later" as anything specified in the license. The license-file itself just doesn't specify the license (as most don't).

https://github.com/LibreCAD/LibreCAD/blob/master/LICENSE isn't GPLv2-only, and
https://github.com/qcad/qcad/blob/master/gpl-3.0.txt isn't GPLv3-only.

In conclusion, if the translations can and should be made GPLv2-or-later, making them compatible with what is in the LibreCAD source repo.

dxli, 2024-05-18:
We should contributors to also assign copyright to LibreCAD.org, so leaving the possibility to reuse their contributions for future relicensed work.

The open source version of the current license is always available, and any relicensing would be free software as well.