LibreCAD/LibreCAD #1593: Maximized window isn't restored correctly on Windows after program re-launch.

Dromantor, 2023-02-14:
I can't really believe nobody noticed this yet, so sorry if there already is an issue.

When I close the program on any of my Windows machines in the maximized state and reopen it the window *fills* the said screen, but it is not properly maximized. It also overlaps a bit to the next screen to the right. See pictures below.

### Expected behavior
Reopen the program, it is maximized again correctly:
[image]

### Observed behavior
Reopen the program, it just fills the screen but isn't maximized (see the maximize button), also overlaps to the right screen and has some space on the left:
[image]
(Yes, these are two different screen sizes, but is also observed with same screen sizes on my other machine.)

### Steps to reproduce or sample file
1. open LibreCAD on Windows
2. click maximize button
3. close program
4. open program again
5. => not maximized anymore, just free window blown up to the screen size and not positioned correctly

### Operating System and LibreCAD version info
Version: 2.2.0
Compiler: GNU GCC 12.2.0
Compiled on: Dec 17 2022
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0) => 22H2 (Build 19045.2486)

lordofbikes, 2023-02-14:
follow-up #1166
not a duplicate, but similar or same code location and worth to consider if somebody takes the challenge

dxli, 2023-07-02:
Commit a6f3423: Qt native window geometry support

closed 2024-05-18 by dxli