LibreCAD/LibreCAD #1594: Won't open in Wayland with error 'qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""'

bmcgonag, 2023-02-15:
### Expected behavior
That the application would open on Ubuntu 22.04 LTS 64 bit default installation.

### Observed behavior
1. In making the Appimage executable, doouble clicking on the application, it never opened. 
2. Running the AppImage from the command line (shell / cli) with the command `./LibreCAD-2.2.0-x86_64_6192c9acf61eca3dca5b40caffd26795.AppImage`, i am presented the following error in the cli stdout:

```
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
execv error: No such file or directory
```
### Steps to reproduce or sample file
1. Download the executable to a default Ubuntu 22.04 64 bit installation.  Ensure packages are up to date.  
2. Make the AppImage executable via the Permissions tab in the Gnome UI, or via the cli.
3. Run the application from the cli using the command given above.

### Operating System and LibreCAD version info
OS: Ubuntu 22.04.1 LTS x86_64 
Kernel: 5.15.0-60-generic 
Uptime: 4 days, 20 hours, 27 mins 
Packages: 1822 (dpkg), 15 (flatpak), 25 (snap) 
Shell: bash 5.1.16 
Resolution: 1920x1080 
DE: Unity <-- I think this is wrong - it's ubuntu's gnome spin.
WM: Mutter 
WM Theme: Adwaita 
Theme: Yaru-red-dark [GTK2/3] 
Icons: Yaru-red [GTK2/3] 
Terminal: Tabby 
CPU: Intel i7-3520M (4) @ 3.600GHz 
GPU: Intel 3rd Gen Core processor Graphics Controller 
Memory: 6272MiB / 15872MiB

Libre CAD Version
2.2.0-x86_64

fuzunspm, 2023-03-13:
same issue on x11 and ubuntu 22.04

nerdCopter, 2023-06-01:
```
$ ./LibreCAD-2.2.0-30-g2a74ef2c-x86_64.AppImage 
execv error: No such file or directory
```

```
$ cat /etc/*releas*
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

nasher87, 2023-06-06:
Same issue in Manjaro
OS: Manjaro Linux x86_64 
Kernel: 6.2.16-2-MANJARO 
DE: GNOME 44.1 
LibreCAD Version
LibreCAD-2.2.1_alpha-62-g4d50e627-x86_64

dxli, 2023-06-07:
I am running Wayland and just tested the Linux images work for me on Gentoo. 

Just wondering whether the issue is due to missing support of the os installation.

Could you install more Wayland support packages and try again? For example, the package xwayland

dxli, 2023-06-25:
Any difference with the latest Linux build? We have updated the building system from Ubuntu 20.04 to 22.04

nasher87, 2023-06-28:
@dxli No, same problem, tested with version LibreCAD-2.2.1_alpha-207-g8f709b91-x86_64, the problem persists
In my case I don't use Ubuntu, I use Manjaro

fa201, 2023-06-28:
LibreCAD-2.2.0-x86_64.AppImage, LibreCAD-2.2.1_alpha-188-g5c4d14de-x86_64.AppImage and LibreCAD-2.2.1_alpha-207-g8f709b91-x86_64.AppImage open correctly with Xubuntu 22.04.LTS (XFCE).

dxli, 2024-05-18:
Closing this one with release candidates available for 2.2.1

closed 2024-05-18 by dxli