LibreCAD/LibreCAD #1595: Bugs or shortcomings noticed

kberkut, 2023-02-16:
1) commands do not work if the program interface language is selected other than English, while English is selected for entering commands
2) it must be done so that when pointing at objects, their visibility changes as in QCAD
3) it is not clear what Revertxdirection does, it does not work even on callout lines with an arrow (Leader)
4) it is impossible to cut a spline line or split into sections
5) the text is not compatible with qcad, it does not display normally turns into crabs when opening a file created in qcad
6) plugins either don't work. Only one works for creating gear

LibreCAD Version: 2.2.0 (appimage)
Compiler: GNU GCC 9.4.0
Compiled on: Dec 17 2022
Qt Version: 5.12.9
Boost Version: 1.75.0
System: Arch Linux

OS: Arch Linux x86_64 
Kernel: 6.1.11-arch1-1 
DE: GNOME 43.2 
WM: Mutter 
WM Theme: Adwaita 
Theme: Adwaita [GTK2/3] 
Icons: Faenza [GTK2/3] 
CPU: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx (8) @ 2.100GHz 
GPU: AMD ATI Radeon Vega Series / Radeon Vega Mobile Series

luzpaz, 2023-05-01:
Better to make a single ticket per issue

dxli, 2023-06-15:
The gear plugin was not of high coding quality.

I am still trying to balance:

1, code quality;
2, solid algorithms;
3, new feature, if the change is maintainable;
4, long term bugs, even if at cost of refactoring, but low risk of regression by a judgement call;
5, code modernization to avoid nonstandard code