LibreCAD/LibreCAD #1598: Saving closed spline to dxf problem

tkvw, 2023-03-17:
Related to https://github.com/mozman/ezdxf/issues/851

[image]

If I save this spline drawing [drawing_spline.zip](https://github.com/LibreCAD/LibreCAD/files/11004931/drawing_spline.zip) it's not formatted correctly in dxf. 

It looks like the spline is saved like an open spline and only the closed flag is set. 

This is how it looks in the Autodesk viewer: 
[image]

lordofbikes, 2023-04-03:
followup https://forum.librecad.org/Closed-spline-bug-with-LibreCAD-tp5722625.html

dxli, 2023-05-26:
The solution:

when saving to dxf: wrap control points: for degree=3, $P_0, P_1, \cdots, P_{n-1}$ becomes $P_0, P_1, \cdots, P_{n-1}, P_0, P_1, P_2$, with 3 points wrapped.

[closed spline](https://pages.mtu.edu/~shene/COURSES/cs3621/NOTES/spline/B-spline/bspline-curve-closed.html)

Within the RS_Spline class, we only wrap control points for evaluation, if no wrapping is detected.

DXF files saved from old LibreCAD versions are expected are not affected.

dxli, 2023-05-30:
Changed to keep control points in LibreCAD always wrapped for closed splines.

Operations need to consider the wrapped control points status
1. adding control points;
2. removing control points;
3. switch between closed and open.

The wrapping happens at the LibreCAD geometry engine side, so the dxfrw side stays ignorant about the wrapping (as befoe).

closed 2023-05-30 by dxli

NathanSweet, 2024-05-08:
Using 2.2.0 and saving a closed spline:
!{}(http://n4te.com/x/10299-g0E6.png)

When viewing it with the [AutoDesk viewer](https://viewer.autodesk.com/) I see it is not closed:
[image]

DXF is [here](https://github.com/LibreCAD/LibreCAD/files/15254552/mirror-large.zip).

NathanSweet, 2024-05-09:
I tried 2.2.1_alpha-164-g09328dc7 (from SourceForge, wasn't found on GH), opened my DXF, saved a new one, and it works!