LibreCAD/LibreCAD #1600: Silent install in Windows opens website after installation

Saxomania, 2023-04-03:
We deploy software to our central managed windows clients via SCCM/MECM and it will be installed via Softwarecenter (user frontend),  this doing installations under SYSTEM account.  

"LibreCAD-Installer-2.2.0.exe /S" installation is working fine but at the the end it opens a website and the default browser opens under SYSTEM account, which is a real problem.

So please add an option to avoid opening this website or, much better, remove this option at all if /S is used.

Coco05051989, 2023-05-02:
Same issue here.

closed 2023-06-04 by dxli

Saxomania, 2023-06-05:
Great, thank you. Just to be sure, this is now working in v2.2.1_alpha-59 but not in the latest stable release 2.2.0?

dongxuli-concord-design, 2023-06-05:
We may want to release 2.2.1.

We got lots of pull requests and lots of other changes since 2.2.0, and some stabilization time is needed.

Saxomania, 2023-06-05:
OK then i will wait for the next stable release. Thanks for your efforts.

PR #1618 (2023-06-04): Add IfSilent jump instruction to skip browser launch after install
Possibly closes #1600

I am not on a Windows machine so this PR is untested, but I don't see a reason why it shouldn't work. The added instruction simply skips over the command to navigate to the website. If it's required I can try spinning up a VM with Windows to really test it out, or maybe someone else can do it on their PC?