LibreCAD/LibreCAD #1603: The path for symbol libraries does not work

kberkut, 2023-04-14:
The path for symbol libraries does not work

### Expected behavior
Displaying custom blocks in the Library Manager

### Observed behavior
The user library is not displayed

### Steps to reproduce or sample file
Add a path for your own block library
[image]

### Operating System and LibreCAD version info

fa201, 2023-04-14:
It works with 
Version: 2.2.0-alpha
Compiler: GNU GCC 11.3.0
Compiled on: Jan  7 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Ubuntu 22.04.2 LTS

Could it be OS dependent ?

dxli, 2023-06-05:
Works for me now.

closed 2023-06-05 by dxli

kberkut, 2024-01-01:
The problem is not in the program, the problem is in the description in the help. It is necessary to add to the help that the settings specify the path to the directory that contains the subdirectory with the dxf files of the symbol library. I suggest adding it to the help:

To connect a third-party symbol library, write the full path to the directory in the LibreCAD settings, for example "/home/user/Library/".
For the specified path, the symbol library files must be located in the subdirectory "/home/user/Library/Library_1". Files with the dxf extension should be located in the "Library_1" subdirectory.
The name of the subdirectory "Library_1" will be displayed in the library manager.