LibreCAD/LibreCAD #1604: Cyrillic text with unicode encoding typed in qcad is displayed incorrectly.

kberkut, 2023-04-14:
Cyrillic text with unicode encoding typed in qcad is displayed incorrectly.

### Expected behavior
Fully readable and editable Cyrillic text.

### Observed behavior
Cyrillic text with unicode encoding typed in qcad is displayed incorrectly.
[image]

### Steps to reproduce or sample file
Create a file in qcad containing Cyrillic text. Open the file in LibreCAD.

### Operating System
OS: Arch Linux x86_64
Kernel: 6.2.10-arch1-1
Resolution: 1920x1080
DE: GNOME 43.4
CPU: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx (8) @ 2.100GHz
GPU: AMD ATI Radeon Vega Series / Radeon Vega Mobile Series
Memory: 5635MiB / 9886MiB

### LibreCAD version info
Версия: 2.2.0
Компилятор: GNU GCC 9.4.0
Собран: Dec 17 2022
Версия Qt: 5.12.9
Версия Boost: 1.75.0
Система: Arch Linux

dxli, 2023-05-23:
Currently, we don't have the ability to detect supported fonts, so please try manual font selection:

# Menu: Tools > Modify > Properties
# Select the text
# In the Font tab, select font (For Cyrillic, "cyrillic-ii", "unicode" and a few others should work).

It's probably not difficult to filter the selection to allow only valid selections for given text.

closed 2023-05-23 by dxli

u2fly, 2023-06-07:
Related https://github.com/LibreCAD/LibreCAD/issues/850