LibreCAD/LibreCAD #1605: Cannot merge splines into a polyline

xmarduel, 2023-04-14:
### Expected behavior
When utilizing the command "create polyline from existing segments" with lets say a line and a spline defined with 3 points,
a single entity resulting of the merge of this line and spline should be created (as for example with a line and an arc)

### Observed behavior
When utilizing the command "create polyline from existing segments" with lets say a line and a spline defined with 3 points,
the two items are not merged into a single entity.

### Steps to reproduce or sample file

### Operating System and LibreCAD version info
Windows. LibreCAD 2.2.0

xmarduel, 2023-05-25:
Hi , thank for the reply.
I would say , if I were the release manager, I would put this „missing feature“ at the top highest priority!

nametabekul, 2024-10-18:
Was this every fixed?

jozsefk9, 2025-07-24:
I would like to have this option available as well. To join splines and lines. For me, on LibreCAD version 2.2.1.2 "create polyline from existing segments" does nothing.

lordofbikes, 2025-07-25:
This ist not possible by DXF definition. The `POLYLINE` entity has not the capcability to keep spline parameters. `POLYLINE` is a list of vertices connected by straight lines or circular arcs.

To achieve this, the spline needs to be converted into an approximation with small line segments or circular arcs. Which does not exist in LibreCAD yet too.