LibreCAD/LibreCAD #1606: Accessibility: icons are invisible in dark themes

Gering-ding-ding-ding-din, 2023-04-14:
### Prologue
Because of health problems regarding my eyes I am forced to use dark/black color themes exclusively combined with a screen brightness of <= 3%. Therefore I'm using a black color theme in KDE Plasma.

### Expected behavior
All icons should be visible.

### Observed behavior
Most icons are invisible.

### Steps to reproduce or sample file
Open LibreCAD in KDE Plasma with a dark theme (e.g. "Breeze Dark" or "Breeze Black") selected.

Breeze Black:
[image]

Breeze Dark:
[image]

### Operating System and LibreCAD version info
Operating System: Arch Linux 
KDE Plasma Version: 5.27.4
KDE Frameworks Version: 5.105.0
Qt Version: 5.15.9
Kernel Version: 6.2.10-arch1-1 (64-bit)
Graphics Platform: X11

LibreCAD Version: 2.2.0-alpha
Compiler: GNU GCC 12.2.0
Compiled on: Dec 17 2022
Qt Version: 5.15.9
Boost Version: 1.80.0
System: Arch Linux

gerardbm, 2023-05-29:
The green of the icons is the only part that have contrast ratio enough to be seen with a dark and a light background. So we only need tot create a version of this icons changing the black for white.

Librecad needs an option to choose between two icons themes, and I will do the icons for the dark mode if someone implements this option.

U--d--o, 2023-10-12:
Same Problem using Dark Mode on MacOS!

There are 2 possibilities, one after the other.

1. the simplest and fast in implementation:
Disable/ignore Dark Mode on MacOS for LibreCAD.

2. the hard way of customization, which can take a little longer after point 1 from me.

**But please, so that you can use LibreCAD also on a Mac in Dark Mode, please implement point 1 asap!**

flywire, 2023-10-12:
Quick and dirty conversion needs testing: https://github.com/flywire/LibreCAD/tree/dark-mode/librecad/res/icons/dark-mode

lordofbikes, 2023-10-13:
Many thanks for this! 👍 
As you have noticed, the icons are built into the binary at compile time.
So this is good for testing, but not practical.
Now it needs code modifications, to load the icons dynamically at run time. Then the user can switch the theme in settings.
Sadly, this can't be done by `sed` 😉

gerardbm, 2023-10-13:
A very interesting implementation of this would be the availability of adding more themes via folders: default, dark, minimal, blued, etc; so they can be switched from a select box. A data file could contain the name, the path...

By default, the dark icons should appear when the user uses a dark UI. If not, the user should be able to specify alternative icons.

@flywire, many thanks for this contribution! I love this!

flywire, 2023-10-14:
Close issue as duplicate of #1144

closed 2024-06-06 by lordofbikes