LibreCAD/LibreCAD #1607: Hatching lines depends on the scale.

kberkut, 2023-04-20:
### Expected behavior
The size of the hatching lines and the rectangle should not change when you change the scale.
The size of the circle hatching should change when changing the hatching settings.

### Observed behavior
The size of the hatching lines and rectangle depends on the scale. This is a bug!
The size of the circle hatching does not change when changing the hatching settings. This is a bug!

### Steps to reproduce or sample file
Draw a rectangle or a line. In the object properties, change the line type to hatching. Change the scale of the drawing display.
Draw a circle. In the object properties, change the line type to hatching. Change the hatching type.
test video
[LC.webm](https://user-images.githubusercontent.com/57869227/233290143-c698baf8-6798-407f-910e-568edafa8a0f.webm)

### Operating System
OS: Arch Linux x86_64
Kernel: 6.2.10-arch1-1
Resolution: 1920x1080
DE: GNOME 43.4
CPU: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx (8) @ 2.100GHz
GPU: AMD ATI Radeon Vega Series / Radeon Vega Mobile Series
Memory: 5635MiB / 9886MiB

### LibreCAD version info
Версия: 2.2.0
Компилятор: GNU GCC 9.4.0
Собран: Dec 17 2022
Версия Qt: 5.12.9
Версия Boost: 1.75.0
Система: Arch Linux

dxli, 2023-05-01:
Conceptually, line type patterns should be in view/paper space, i.e. scaling shouldn't change pattern sizes.

dxli, 2023-05-24:
harder than expected. Still working on.