LibreCAD/LibreCAD #1609: Allow \F in Text Field

jmj999, 2023-05-10:
Hi,

I am used to put in a text field the path of my drawings.

Today I work on  C:\Users\zc01w\OneDrive\JMJ\CFAO\_RealisationJMJ\2023_MDP\FrigoSMEG\Dessins

I am unable to print" \F"  : nor the "\" nor the "F" are printed.

An idea (except rename the project :-)    )? 

Regards
JMJ

lordofbikes, 2023-05-10:
See #1286
There are several links explaining the use of formatting codes, which are initiated with a `'\'`.
`\F` is for font selection. I'm not sure about their implementation in LiberCAD, but this is probably the cause for your issue.

To use `\` in text you should double each occurrence to escape the `\`.
E.g. `C:\\Users\\zc01w\\OneDrive\\JMJ\\CFAO_RealisationJMJ\\2023_MDP\\FrigoSMEG\\Dessins`.
Haven't tested it, but when `\F` is interpteted, `\\` should be too and result in a single `\` in the end.

jmj999, 2023-05-10:
Hi lordofbikes,

I never had to escape the \ until today with the \F sequence( first project starting with a F)

\\F prints the \ but not the F
\\ F  ( 2 backslashes and a space before the F) prints \\ F ( 2 backslashes and a space before the F)

\F seems to always be interpreted (no way to escape the \) 

regards
JMJ

dxli, 2023-05-10:
This appears to be a bug in MText.

\F{ } specifies font change.

The bug is when \F is not followed by '{'.

dxli, 2023-05-10:
added a fix:
https://github.com/LibreCAD/LibreCAD/commit/03f0cd27bf4841ecdba595a75a0a10209d759f14

jmj999, 2023-05-10:
Thanks,
will wait for the fix... until it is fixed, I rename the project.
regards
JMJ

closed 2023-05-10 by jmj999