LibreCAD/LibreCAD #1610: App stops responding after zooming in/out to a specific zone

mich-j, 2023-05-11:
### Expected behavior
LibreCAD allows to zoom in and out to every part of a drawing.

### Observed behavior
LibreCAD stops responding, when I zoom in and zoom out a few times to a specific zone in my drawing. File that I'm working on is a section cut from an assembly drawing, created in [FreeCAD TechDraw Workbench](https://wiki.freecad.org/TechDraw_Workbench) and exported to .dxf. I deleted by "trial and error" lines which did not cause any freezing, which left me with a something like minimal working example, which is included in this issue ticket (`drawing.zip`). Unfortunately, I am not able to pinpoint which lines exactly caused this problem.

I had an opportunity to import this file to [GStarCAD](https://www.gstarcad.net/) on someone else's PC and it indeed prompted that some of the lines are broken. It said that it repaired them, but did not inform which ones were faulty. I imported this 'repaired' file to LibreCAD again, but it unfortunately did not help in the freezing issue.

Linking similar issues: #1000, #891 

### Steps to reproduce or sample file
[drawing.zip](https://github.com/LibreCAD/LibreCAD/files/11455649/drawing.zip)

### Operating System and LibreCAD version info

- System: Ubuntu 22.04.2 LTS x86_64 

- FreeCAD: I tried two versions:
  - **release**
  Version: 2.2.0
  Compiler: GNU GCC 9.4.0
  Compiled on: Dec 17 2022
  Qt Version: 5.12.9
  Boost Version: 1.75.0
  - **built from source**
  Version: 2.2.0-15-g14816bd5
  Compiler: GNU GCC 11.3.0
  Compiled on: May 11 2023
  Qt Version: 5.15.3
  Boost Version: 1.74.0

Both crashed in a similar manner.

dxli, 2023-05-19:
This is a problem of splines. I will be working on this one.

dxli, 2023-05-20:
Some background on the math part: 
[Quadratic Spline Curve Length](https://members.loria.fr/SHornus/quadratic-arc-length.html#:~:text=The%20length%20of%20the%20quadratic,(%20t%20)%20%7C%20d%20t%20.)

We are using a dimensionless quadratic function
$$\int 2\sqrt{1+v^2}dv=v\sqrt{1+v^2}+ \ln(v+\sqrt{1+v^2})$$

The issue is due to negative v, when $v<-10^{-10}$, when $v$ and $\sqrt{1+v^2}$ are close in absolute values but of different signs, and $v + \sqrt{1+v^2}$ gets unstable numerically with loss significant digits.

The fix, as usual, avoid summation floating points of different signs, for negative $v$,
$$v+\sqrt{1+v^2} = \frac{1}{\sqrt{1+v^2}-v}$$

mich-j, 2023-05-21:
With your commits, on LibreCAD:
```
Version: 2.2.1_alpha-9-g8c2e3ce6
Compiler: GNU GCC 11.3.0
Compiled on: May 21 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Ubuntu 22.04.2 LTS
```
I observed no freezing on the sample file. Also I tested [BancadaCozinha-Vistas.dxf.gz](https://github.com/LibreCAD/LibreCAD/files/2249060/BancadaCozinha-Vistas.dxf.gz) file from #1000 and it also turned out OK.

Unfortunately on file [Transistors.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/1017101/Transistors.dxf.zip) from #891 freezing was still observed.

So @dxli, I think that this issue and #1000 can be closed. I leave this decision for you. If you need more testing, I can help out as much as I can.

dongxuli-concord-design, 2023-05-21:
Let's fix related issues together here. The quadratic spline was introduced by Pavel long time ago, and it's a wonderful ideal to allow our analytical treatment of quadratic curves to be extended to splines (quadratic).
Numerical robustness is always hard for any CAD system. For example, finding coincident points with rounding off errors is essentially a k-cluster problem, therefore NP-hard.

dxli, 2023-05-22:
#891 is a separate issue specific for polyline bulge handling.

#1000 is a duplicate of this one due to SplinePoints length numerical stability.

Successfully tested.

closed 2023-05-22 by dxli