LibreCAD/LibreCAD #1614: Feature Request: Measure > Point Coordinate (aka Tools > Info > Point Coordinate)

bernd-wechner, 2023-05-25:
Related to https://github.com/LibreCAD/LibreCAD/issues/359 (but I feel that one is a little less specific or clear).

There is a naming inconsistency here BTW, on the toolbar there is a Measure group (as gauged by the pop-up ToolTip) and on the Tools menu there is an Info submenu that is identical (to wit a Measure-Info inconsistency).

This group could/should IMHO have a "Point Coordinate" entry, which just prints to console points as they are selected.

Easy to add, as the "Measure > Polygon Area" tool already does this, it just takes some experimenting to find that work around and to ignore the masses of Area prints you don't want is all and work it out - i.e. it isn't intuitive or the intended use of that tool.

I find myself often wanting to coordinates of selected points, when compiling console commands. And I am using this work around but I feel that a dedicated tool is warranted as a) it took me some while to work out I could use the Polygon Tool for this and b) It's very noisy and not a nice tool for the job (I just want the point coordinates).

dellus, 2023-05-25:
If you click the same point twice you don't get the area and circumference info.

dellus, 2023-05-25:
Or, even better, click once and cancel the operation with right mouse click, then just go to the next point, as the command still is active, and so on. Then you have single entries for each point in the command line area.

bernd-wechner, 2023-05-27:
Thanks for the tips. Still think this is a feature worth implementing as it would far more discoverable and can present coords in the comma separated format needed for the commands one would construct using point measures. It would be great to just click them and get a list of the coords and build a command line instruction from that.

sand1024, 2024-10-26:
@bernd-wechner  - please check the latest alpha (https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest)
There is a PenPallette widget (in widgets menu) that allows to collect coordinates. If it addresses your request - please close the issue or provide your feedback, if any. 
Thank you.

sand1024, 2026-09-26:
supported by Entities Info and properties widgets

closed 2026-09-26 by sand1024