LibreCAD/LibreCAD #1616: copy and paste blocks with rotation [Strg+C -> Strg+V] unexpected behaviour

tatrapikao, 2023-05-28:
### Expected behavior

a duplicate of the selected block is created

### Observed behavior

a duplicate of the selected block is created but with different rotation. Using the Move/Copy function instead works as intended.

### Steps to reproduce or sample file

insert a block
rotate it
copy and insert via [Strg+c] -> [Strg+v]

### Operating System and LibreCAD version info

Version: 2.2.0-2-gf55f341e
Compiler: GNU GCC 9.4.0
Erstellt am: Dec 19 2022
Qt Version: 5.12.9
Boost Version: 1.75.0
System: Ubuntu 22.04.2 LTS
[image]

dxli, 2023-06-02:
The apparent cause

1, at action copy, the block and its entities are copied to clipboard with all entities rotated by the insertion angle;

2, at action paste, the block and its entities in the clipboard is copied to the document, moved to the new insertion point and rotated by the block angle.

The symptom: the entities are rotated twice.

dxli, 2023-06-03:
This is likely a regression introduced by d6d5b037ccfd02abd0bf8623c233535d29e9b808

gerardbm, 2023-06-04:
This seems related with the pull request #1498, which should be closed if this issue is fixed. Let me now if I'm wrong, so I can delete the reference from this comment.

dxli, 2023-06-05:
Yes, the same issue as pull request #1498. An actual fix is still needed to keep the insertion angle.
Let's keep both open.