LibreCAD/LibreCAD #1624: Latest CI-built AppImage can not locate resources paths

u2fly, 2023-06-08:
### Expected behavior

LibreCAD AppImage could load built-in resources(fonts, locales, etc.)

### Observed behavior

Since `LibreCAD-2.2.1_alpha-125-ga28f13b5-x86_64.AppImage` all next AppImage builds can not locate built-in resources.

The last working AppImage (wihtout issue): `LibreCAD-2.2.1_alpha-85-g2d000d15-x86_64.AppImage`

### Steps to reproduce or sample file

1. Download latest CI built AppImage;
2. Try to set interface locale or create text and try to select font.

### Operating System and LibreCAD version info

- OS: Linux / Debian 11
- Version: `LibreCAD-2.2.1_alpha-127-gc9ec48cf-x86_64.AppImage`

### Related issues

- https://github.com/LibreCAD/LibreCAD/issues/1488

dxli, 2023-06-12:
Fixed by 09328dc

closed 2023-06-12 by dxli

u2fly, 2023-06-13:
> Fixed by 09328dc

@dxli, Nope, issue still exists - it still impossible to set locale, choose font for text elements or choose pattern (all these drop-down lists are empty).

Actual latest `LibreCAD-2.2.1_alpha-164-g09328dc7-x86_64.AppImage` still does not reconize built-in resources.