LibreCAD/LibreCAD #1630: I can't compile LibreCAD after last update

gerardbm, 2023-06-10:
### Expected behavior

### Observed behavior

This is the error I'm getting after the command `make -j4`:

```
[...]
Makefile:5710: recipe for target '../../generated/librecad/obj/rs_ellipse.o' failed
make[2]: *** [../../generated/librecad/obj/rs_ellipse.o] Error 1
make[2]: *** Waiting for unfinished jobs....
lib/engine/lc_splinepoints.cpp: In function ‘double DrawPatternLine(const std::vector<double>&, int, double, QPainterPath&, RS_Vector&, RS_Vector&)’:
lib/engine/lc_splinepoints.cpp:1554:16: warning: unused variable ‘dt2bak’ [-Wunused-variable]
         double dt2bak=dt1;
                ^~~~~~
make[2]: Leaving directory '/home/gerard/Workspace/varis/librecad/librecad/src'
Makefile:42: recipe for target 'sub-src-make_first' failed
make[1]: *** [sub-src-make_first] Error 2
make[1]: Leaving directory '/home/gerard/Workspace/varis/librecad/librecad'
Makefile:88: recipe for target 'sub-librecad-make_first-ordered' failed
make: *** [sub-librecad-make_first-ordered] Error 2
```

This is thet last part of the error, because it's too long.

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

fa201, 2023-06-10:
Which OS ?

gerardbm, 2023-06-10:
Debian GNU/Linux 9 (Stretch)

dxli, 2023-06-10:
To see the actual error message, run "make -j1" after the building failure.

gerardbm, 2023-06-10:
I redirected it to a file with `2>`

[log.txt](https://github.com/LibreCAD/LibreCAD/files/11712421/log.txt)

dxli, 2023-06-11:
the log file contains no compiler command lines.

gerardbm, 2023-06-11:
I don't know what I should do. I always do the following: `qmake -r` and then `make -j4`. At the end of this process, the binary file `./unix/librecad` is not created. The output is so long that I don't know how to copy it. Let me show you a video with the process: [Youtube](https://www.youtube.com/watch?v=EktxwhTpm3g)

gerardbm, 2023-06-11:
Well, I'm gonna use the AppImage from now. Probably this issue occurs because my OS is too old. Close it if it's the case. Thanks a lot for the fixes and the improvements.

dongxuli-concord-design, 2023-06-11:
No, your compilern new enough, and it's just one incorrect compiler command line option. Try to supply a CXXFLAGS="-std=c++1y" in custom.pro

gerardbm, 2023-06-13:
Is it possible that the compiler is not reading the file `custom.pro`?

dongxuli-concord-design, 2023-06-13:
No.

librecad.pro is our entry point for qmake.

custom.pro is called by librecad.pro

custom.pri is used by librecad/src/src.pro

gerardbm, 2023-06-13:
Finally I could compile LibreCAD, adding the following line in `custom.pro`:

```
QMAKE_CXXFLAGS += -fext-numeric-literals
```

Now I have a binary file. However, when I execute it I get the following output error:

```
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
pure virtual method called
terminate called without an active exception
```

dxli, 2023-06-13:
Could you update your source and try again?

It's probably due to a mistake I made today. Should be fixed in master already.

gerardbm, 2023-06-13:
Yes, that worked perfect now! Thanks a lot again.

closed 2023-06-13 by gerardbm