LibreCAD/LibreCAD #1631: "Free snap" needs a deep review

gerardbm, 2023-06-10:
### Expected behavior

When the "Free Snap" is enabled, it should allow the free movement until the mouse is close to a snap area, then it should snap to the marked areas (let's say the end of a line, or the center of a circle). 

### Observed behavior

The expected behavior is ok at startup, but it stops working after drawing some elements (lines or circles). So the behavior is not consistent. I'm not able to determine exactly when it breaks the expected behavior, but it happened twice after I created 3 circles and I added a fourth circle tangencial to 3. The 4th circle still have the highlighted color after ending the action. From this moment, I can see the "Free snap" does not allow to snap other selected points.

I also observed that a saved file contains the error itself, so I upload the file.

### Steps to reproduce or sample file

File uploaded: 
[aaa.zip](https://github.com/LibreCAD/LibreCAD/files/11711988/aaa.zip)

### Operating System and LibreCAD version info

Version: 2.2.1_alpha-137-ge40b4569
Compiler: GNU GCC 6.3.0
Compiled on: Jun  9 2023
Qt Version: 5.7.1
Boost Version: 1.62.0
System: Debian GNU/Linux 9 (stretch)

gerardbm, 2023-06-12:
That works better. 

However, about the consistency of the snapping area: I can see that the snapping area changes depending on the visibility of the grid. If the grid is visible, the snapping area is smaller. If the grid is not visible, the snapping area is bigger. I think that this shouldn't be dependent of the grid visibility. This should be dependent of the viewport, so you can zoom in or zoom out, and have always the same snapping area, with independence of the grid visibility.

dxli, 2023-06-12:
One issue, if the free snapping distance is larger than half of the grid size, free snapping is not possible with grid on.

That's the motivation of making the snapping distance be aware of the grid status and size.

closed 2026-09-26 by sand1024