LibreCAD/LibreCAD #1633: Can open a DXF file after last compile

fa201, 2023-06-11:
### Expected behavior
Open a LibreCAD valid DXF works.
### Observed behavior
Message: `Import Error: undefined error`
### Steps to reproduce or sample file
Open any DXF file generated by LibreCAD.
### Operating System and LibreCAD version info
Version: 2.2.1_alpha-150-g97c00660
Compiler: GNU GCC 11.3.0
Compiled on: Jun 11 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Ubuntu 22.04.2 LTS

fa201, 2023-06-11:
A new DXF file generated by the same version of LibreCAD can be saved with a new layer (non 0) and a rectangle. When opening the file, there is no error message but no new layer and no rectangle.

fa201, 2023-06-11:
Same issue with which was working a few month ago:
Version: 2.2.0-alpha
Compiler: GNU GCC 11.3.0
Compiled on: Jan  7 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Ubuntu 22.04.2 LTS

Could it be that DXF library is broken ?

dxli, 2023-06-11:
81d93c7f192f00ea845cc8171f8bc75944958bd1

issues #1632/1633 : CI linux build fix
    
    AppImageKit: image building failure due to missing of Android emulators
    under /usr/local/lib/android

    The fix: remove /usr/local/lib/android before running appimagetool

fa201, 2023-06-11:
I do not understand

fa201, 2023-06-11:
There is no such `android` folder on my system. I compile LibreCAD and as far as I know, I do not run appimagetool.

dxli, 2023-06-11:
Sorry, the fix is for CI: Continuous Integration.

If you mean your local build, could you also try the latest CI build?

Does it happen to all dxf files? any debugging messages from dxf file opening failure?

fa201, 2023-06-11:
Yes it happens on all DXF files (all generated by LibreCAD). There is no other messages apart from the one I shared previously.

I do not know to use the CI build.

dxli, 2023-06-11:
For CI builds https://github.com/LibreCAD/LibreCAD/releases

fa201, 2023-06-12:
Same issue with `-d6` my buffer is not big enough to hold all the messages.

Without any option, I have the following log.

[message_app_image.txt](https://github.com/LibreCAD/LibreCAD/files/11727673/message_app_image.txt)

dxli, 2023-06-12:
Better to have the complete log, like

```bash
./LibreCAD-2.2.1_alpha-156-g9ad3a64f-x86_64.AppImage -d6 &> librecad-log-$(date +"%s").txt
```

fa201, 2023-06-13:
[librecad-log-1686683964.txt](https://github.com/LibreCAD/LibreCAD/files/11738454/librecad-log-1686683964.txt)

dxli, 2023-06-15:
Can you open an empty dxf?
Then, a dxf with a single line;
Followed by a dxf with a single block which contains a single line.

Your log file shows a very complex dxf file and without the actual error message.

fa201, 2023-06-15:
[log.zip](https://github.com/LibreCAD/LibreCAD/files/11762479/log.zip)
I attached also the test files.

fa201, 2023-06-21:
Using continuous CI AppImage and my last build, it works now. 
I discovered that default format for opening file was `QCAD 1.x *.dxf` for an unknown reason. Changing to `LibreCAD DXF`, I can open my file.
The issue could be an error at my side so I prefer closing the issue. 
Sorry for any extra work.

closed 2023-06-21 by fa201